openrazer/openrazer #1255: Support for [device name here]

Bryiu, 2020-09-15:
### Details about the problem

I have to type this on my phone because the keyboard is currently bricked. Doesnt light up, when i press a key it holds that key for a long time.
example: ffffffffffffffffffffffffffffffffffffffffffffffff

that was one key press

verbose output:
```
Verbose enabled
/usr/bin/polychromatic-controller:290: DeprecationWarning: Gtk.Window.set_wmclass is deprecated
  self.window.set_wmclass("polychromatic-controller", "polychromatic-controller")
/usr/bin/polychromatic-controller:292: DeprecationWarning: Gtk.Widget.modify_bg is deprecated
  self.window.modify_bg(Gtk.StateType(0), Gdk.Color(0, 0, 0))
Set scale to: 1.0
Initialised Window.
Version 0.3.12-stable
Loading daemon...
Gathering devices...
 -- Found: Razer BlackWidow 2019
 -- Found: Razer DeathAdder Elite
Loading application profiles...
```

### Operating System / Desktop Environment

Ubuntu 20.04 LTS

### Affected Devices

Blackwidow 2019

this was copied over from my post on polychromatic

z3ntu, 2020-09-15:
Can you try uninstalling openrazer and see if the problem persists?

Bryiu, 2020-09-15:
Yes, the problem still persists.

Bryiu, 2020-09-15:
After uninstalling Openrazer, unplugging and replugging the keyboard, it now works.

Thanks!

closed 2020-09-15 by Bryiu