openrazer/openrazer #1258: Please add support for DeathAdder V2 Mini

setlord, 2020-09-16:
### Device Info

* Name: DEATHADDER V2 MINI
* Product Number: RZ01-03340100
* Model Number: RZ01-0334 Gaming Mouse

### Device Information
```
Bus 001 Device 019: ID 1532:008c Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x008c 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer DeathAdder V2 Mini
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx 1 root root  10 sep 15 23:18 usb-04d9_USB_Keyboard-event-if01 -> ../event11
lrwxrwxrwx 1 root root  10 sep 15 23:18 usb-04d9_USB_Keyboard-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 sep 14 23:46 usb-ASUSTeK_Computer_Inc._N-KEY_Device-event-if02 -> ../event15
lrwxrwxrwx 1 root root  10 sep 14 23:46 usb-ASUSTeK_Computer_Inc._N-KEY_Device-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 sep 14 23:46 usb-ASUSTeK_Computer_Inc._N-KEY_Device-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root  10 sep 14 23:46 usb-Azurewave_USB2.0_HD_UVC_WebCam_0x0001-event-if00 -> ../event19
lrwxrwxrwx 1 root root   9 sep 14 23:46 usb-Jieli_Technology_USB_PHY_2.0-event-if00 -> ../event9
lrwxrwxrwx 1 root root  10 sep 16 14:59 usb-Razer_Razer_DeathAdder_V2_Mini-event-if01 -> ../event30
lrwxrwxrwx 1 root root   9 sep 16 14:59 usb-Razer_Razer_DeathAdder_V2_Mini-event-mouse -> ../event6
lrwxrwxrwx 1 root root  10 sep 16 14:59 usb-Razer_Razer_DeathAdder_V2_Mini-if01-event-kbd -> ../event27
lrwxrwxrwx 1 root root  10 sep 16 14:59 usb-Razer_Razer_DeathAdder_V2_Mini-if02-event-kbd -> ../event31
lrwxrwxrwx 1 root root   9 sep 16 14:59 usb-Razer_Razer_DeathAdder_V2_Mini-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
I can't undertand what to do

### Synapse Screenshots
[image]
[image]
[image]
[image]
[image]

zeroedout, 2020-10-28:
### My Device:
Output of `lsusb -d 1532: -v`
[lsdevoutput.txt](https://github.com/openrazer/openrazer/files/5455071/lsdevoutput.txt)

Output of `ls -lah /dev/input/by-id/`
[lsusboutput2.txt](https://github.com/openrazer/openrazer/files/5455072/lsusboutput2.txt)

**Model Number** RZ01-0334
**Product Number** RZ01-03340100

### Packet Captures
I've created some packet captures from the Synapse app. These are uncommented so I can redo them with comments on when the feature change is executed. Please let me know if you need more information or different captures. The filter I used:
 `usb.bus_id == 2 && usb.device_address == 8` 

The device as reported by lsusb `Bus 002 Device 008: ID 1532:008c Razer USA, Ltd Razer DeathAdder V2 Mini`

The packet captures are uploaded here https://drive.google.com/drive/folders/15DjOZs4CwQKXFQC6F0TtlJd4q8CD0i0x?usp=sharing

I'll make captures of different features (commented) if someone can work with these.

sonnyarvin, 2020-11-17:
Bus 001 Device 006: ID 1532:008c Razer USA, Ltd

device not recognize 

[image]

zeroedout, 2020-11-18:
That is correct @sonnyarvin. Setlord documented the config, I've provided some USB packet captures. Now someone has to write the driver and submit the pull request to the project. I'm a terrible programmer but I've taken a look at the source code and think I understand the structure. I may try to make a simple driver with one of the features to see if I can get the hang of it. Perhaps (hopefully) someone else with stronger programming skills will write one faster than I can.

z3ntu, 2020-11-18:
@zeroedout Here are a few pointers from the packet captures for what you need:
* brightness https://github.com/openrazer/openrazer/blob/03df9f3bde2a3e7e47153c2e142d942da9d68564/driver/razermouse_driver.c#L1047
* custom frame: https://github.com/openrazer/openrazer/blob/03df9f3bde2a3e7e47153c2e142d942da9d68564/driver/razermouse_driver.c#L544
* `MATRIX_DIMS = [1, 1]` in daemon
* dpi and poll rate are normal
* dpi stages from #1138 (will merge shortly) are supported
* effects are not visible in the pcaps, this will be trial and error for you but you can orient yourself with the mamba elite (as custom frame & brightness is the same as on this device)

z3ntu, 2022-10-31:
This is solved since #1368, closing.

closed 2022-10-31 by z3ntu