openrazer/openrazer #1259: Razer Huntsman profile saving

athakkar99, 2020-09-17:
Hey, I'm pretty new to the openrazer scene, and I just had a few questions. I just got a razer huntsman, and I, like a lot of other people, was unable to save my lighting profiles to the keyboard's onboard memory. After a lot of digging for a workaround on the razer forums, someone referred me here, and said that they were able to save a lighting profile on their huntsman keyboard with this software. If I save an RGB profile on my keyboard using openrazer, and then unplug it and plug it in to a different device, will it keep the saved rgb profile? 

### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

z3ntu, 2020-09-20:
OpenRazer doesn't support the "onboard memory" yet - we haven't figured out the parameters of the command sent to the device

> Class | CMD | Notes
> -- | -- | --
> 0x05 | 0x02 | probably set current onboard profile with param index
> 0x05 | 0x08 | set onboard profile, first param is index probably and the rest unknown

from https://github.com/openrazer/openrazer/wiki/Unknown-commands

But generally the "old school way" of saving the effect to the device is by telling the device to store the effect we set - e.g. https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L1147 the "VARSTORE" parameter (instead of NOSTORE) which sets the "profile" as Razer seems to call it internally and it remembers the effect for the next time you power on the device (a.k.a plug it in somewhere). So generally, yes if you set e.g. the wave effect and you plug the device into a different computer this effect should still be active - until you change it. Any further questions?

athakkar99, 2020-09-21:
Hey, 

First of all, I just want to say thank you for the incredibly detailed and prompt answer; I've never seen a Github this polished or with this strong of a community. Also, I'm not the savviest when it comes to tech (I took AP computer science in high school, so I know the VERY basics, but nothing much more.) If you could give me the ELIF on how to do it (explain like im five), that'd be incredible. I've seen this particular issue on tons of forums, so I'm sure that if this was easier to find/use a large portion of Huntsman and other synapse 3 only users would be inclined to use OpenRazer and its utilities. I'm sure you're very busy, but I'd greatly appreciate if you could drop a step by step whenever you have time, and I'll definitely do it.

z3ntu, 2020-09-26:
Figuring out what the parameters are means doing a bunch of usb packet captures of Synapse doing these commands and correlating what you set to what bytes have been sent over the wire which will be tricky. I can't really help much there though.

closed 2021-06-20 by github-actions[bot]