openrazer/openrazer #1260: The colon in the key name makes apt key fail

Stig124, 2020-09-18:
### Distribution

Debian 10

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I actually encountered that issue before => fish-shell/fish-shell#7134
The colon in the `hardware:razer` GPG key (in `/etc/apt/trusted.gpg.d`) makes apt key fail and you need to replace manually the colon by an hypen to make apt key (even if it's actually deprecated) run

It should have already been upstream-fixed by the openSUSE maintainers but it hasn't been working => openSUSE/software-o-o#843
I already opened an issue at openSUSE/software-o-o#869

z3ntu, 2020-09-20:
Thanks for opening an issue (and patching it?) upstream. Unfortunately there's nothing we can do on our side but thanks for informing us.

z3ntu, 2020-11-09:
Site was deployed now.

closed 2020-11-09 by z3ntu