openrazer/openrazer #1263: Razer Lancehead TE White - no control over side leds.

hypixus, 2020-09-22:
### Distribution

2.8.0 - bundled with Polychromatic on AUR

PC: Manjaro aarch64 on Raspberry Pi 4B, Manjaro x86_64 on Fujitsu v6505, both kernel ver 5.4

Device: Razer Lancehead Tournament Edition (Mercury White)

### Support

- [ yes ] Have you tried rebooting?
- [ yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm only able to change Razer logo light and scroll wheel.
Side lights are not possible to change, however changing LED brightness of entire device works on those.
Multiple reboots, changing kernel versions, clean system installation does not change it.

lah7, 2020-09-22:
Duplicate of #1111.

There's nothing wrong with your installation of OpenRazer, but rather the Polychromatic frontend is a bit behind right now.

If you wish to continue with Polychromatic v0.3.12, [there is a branch by **nightsky30**](https://github.com/nightsky30/polychromatic/tree/side-leds) you can try that adds side LEDs - that is, until a new version in development catches up (https://github.com/polychromatic/polychromatic/issues/263)

Or, you can try [RazerGenie](https://github.com/z3ntu/RazerGenie) which should support side LEDs. 

Or, you can interface with the OpenRazer daemon directly with this one line command:

    python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; a.fx.misc.left.static(0,255,0); a.fx.misc.left.brightness = 100; a.fx.misc.right.static(0,255,0); a.fx.misc.right.brightness = 100; a.fx.misc.logo.static(0,255,0); a.fx.misc.logo.brightness = 100; a.fx.misc.scroll_wheel.static(0,255,0); a.fx.misc.scroll_wheel.brightness = 100;"

_This will set everything to static green (0,255,0) at 100% brightness._

hypixus, 2020-09-22:
Thank you very much for very detailed answer - all I really wished for was to set all lights to static white, and this command is easily modifiable to my liking. Closing.

closed 2020-09-22 by hypixus