openrazer/openrazer #1264: Support terraria

rubiksmaster02, 2020-09-24:
Terraria came out with a razer chroma support update along time ago, but openrazer doesn't support terraria making it a bi disappointment.

lah7, 2020-09-24:
This would be the "Razer Chroma SDK" or ["Works with Razer Chroma"](https://developer.razer.com/works-with-chroma/) program which is proprietary to Razer and only works on Windows and Synapse 3. Therefore, OpenRazer cannot support this for native Linux games.

In theory, if the Windows version of a game is to be ran under Wine/Proton, someone could write a DLL override to translate the calls to the Chroma SDK to OpenRazer's library instead - but I'm not aware how easy this is or [the legalities](https://developer.razer.com/sdk-terms-and-conditions/) (if any).

CalcProgrammer1, 2020-09-24:
The Aurora people have a way to extract the Chroma SDK color data from the official Razer Chroma SDK DLL, which is necessary as some games signature-check the DLL.  Unfortunately, Aurora only works on Windows as it is C# using Windows Forms.  I tried it in Wine, it just crashes during startup.  On Windows, I am able to run all my devices through my OpenRGB software which uses a Windows DLL version of OpenRazer to control my Razer gear.  The signal path looks like this:

Overwatch (a Chroma SDK game) --> Razer Chroma SDK DLL --> Aurora --> OpenRGB --> OpenRazer-win32 --> Razer hardware

I'd like to get something working in Linux as well.  It's possible that Artemis 2 will support this Chroma SDK integration in the future and Artemis 2 does work in Wine (albeit with some flickering UI issues).  I've already confirmed I can control my RGB devices with Artemis 2 on Linux, and there was a WIP plugin for Overwatch demoed on the Artemis discord, so it might be possible in the (hopefully near) future.

rubiksmaster02, 2020-09-24:
Is it possible to script openrazer from c#?

CalcProgrammer1, 2020-09-24:
Sure, you can control OpenRazer several different ways.  I personally do not use the OpenRazer Python daemon (which I believe uses d-bus for communication) so I'm not familiar with interacting with it.  I use the low level components of OpenRazer in my own RGB control app OpenRGB, which also supports a bunch of other RGB devices from various vendors.  OpenRGB provides a network-based SDK.  One of the Aurora developers wrote a C# interface to OpenRGB which he then used to write integrations for Aurora and Artemis.

rubiksmaster02, 2020-09-24:
I have no idea of what half of that stuff is heh

rubiksmaster02, 2020-09-24:
So is the only important stuff the stuff in /driver ?

rubiksmaster02, 2020-09-24:
Like if i was to start a c# project that utilizes open razer, would i only have to include the files in /driver?

rubiksmaster02, 2020-09-24:
Also, Aurora is windows and it doesnt use openrazer

dasizeman, 2020-09-24:
@rubiksmaster02 Please don't do multi-posting like this, it creates a lot of spam for people who are watching the issue.

CalcProgrammer1, 2020-09-24:
Aurora itself doesn't use OpenRazer.  My project OpenRGB (https://gitlab.com/CalcProgrammer1/OpenRGB) uses OpenRazer, and it supports both Linux and Windows.  I use a wrapper that builds OpenRazer as a DLL for Windows (https://github.com/CalcProgrammer1/openrazer-win32).  Aurora's development branch can control OpenRGB, and OpenRGB can control OpenRazer.

If you want to go this route, look into this C# project: https://github.com/diogotr7/OpenRGB.NET

I don't know if it builds on Linux or not, I'm not a C# programmer myself other than dabbling a bit in the Aurora codebase to map LEDs for OpenRGB.

Otherwise, if you want to talk to OpenRazer without using OpenRGB, you'll have to look into the documentation for OpenRazer's daemon and what interfaces apps can use to control it.  Like I said, I think it uses D-bus, another thing I'm not all that familiar with.  You'd have to write your own C# library to communicate with that as I don't think one currently exists.

rubiksmaster02, 2020-09-24:
I was thinking either, i was going to rewrite the drivers in c#, or just wrap the c++ ones

hacker1024, 2022-07-12:
I've started working on an implementation of the official REST API in Dart [over here](https://github.com/Razer-Chroma-Dart/razer_chroma.dart/tree/master/razer_chroma_rest_server). It's been a while since I last touched it, though.

Also in that repo is a (out-of-date, missing useful Dart FFI improvements) macOS platform implementation. It uses the [librazermacos](https://github.com/1kc/librazermacos) library, which is actually a port of the openrazer driver to macOS. It might be able to be ported to openrazer pretty easily - though I'm not sure if using the driver directly is supported in openrazer?

lah7, 2024-03-19:
#2224 reminded me of this issue:

I guess _in theory_, if I understood the [Razer Chroma SDK](https://developer.razer.com/works-with-chroma/download/) correctly, it's some kind of ReSTful web server running in the background, with supporting applications/games connecting to it via one of these libraries:

* [Razer Chroma SDK C++](https://assets.razerzone.com/dev_portal/C%2B%2B/en/index.html)
* [Razer Chroma SDK REST](https://assets.razerzone.com/dev_portal/REST/html/index.html)
* [Razer Chroma SDK Websocket](https://assets.razerzone.com/dev_portal/websocket/html/index.html)

_In theory,_ it just needs the server part "clean-room reverse engineered" and adapted to work with OpenRazer so these apps/games have a server mimicking the official SDK. Since OpenRazer provides a Python library, Python would be an easy choice. 

That said, such compatibility might **only work** for Windows/Wine/Proton versions of games. Looks unlikely that Linux native binaries embed the official C++ library based on [this](https://assets.razerzone.com/dev_portal/REST/html/_rz_errors_8h.html) reference to the Windows SDK. [Euro Truck Simulator 2](https://blog.scssoft.com/2019/05/razer.html) and [Worms W.M.D.](https://www.facebook.com/razer/videos/worms-wmd-official-razer-chroma-integration/10155904936397576/) are two games I know that have Razer Chroma SDK support, but their Linux binaries don't have any `.so` files (DLL-equivalent) named like it nor is it listed in `ldd`. ETS2 (Linux) didn't make any HTTP requests made to [ports 54235 or 54236](https://assets.razerzone.com/dev_portal/REST/html/index.html).

I'll leave this issue open with a "help wanted" in case anyone is interested in a challenge; and so users can discover the technical reasons.

CalcProgrammer1, 2024-03-19:
The Artemis and Aurora projects have already done the work of reverse engineering the Chroma SDK.  These projects can read Chroma lighting from Razer Chroma supporting games and control lights on third party devices, though only on Windows AFAIK.  Artemis supports Linux but my past attempts to use the Chroma capture feature only worked on Windows (albeit it's been a while).  From what I remember, it actually uses a shared memory buffer to pass the lighting control information from the game (Razer Chroma SDK DLL) into the Chroma SDK background service that normally controls the lighting.

CalcProgrammer1, 2024-03-19:
Maybe the shared memory buffer is updated by the Chroma background service and not the DLL itself.  I do vaguely remember needing one of the Chroma SDK services running in order for the Aurora chroma integration to work.

I8aDOG, 2024-10-09:
Any news on this? Would love to play games like the new Silent Hill remake with the contextual color animations on Linux.