openrazer/openrazer #1265: Monitor screens go blank. Page flip error.

malucious81, 2020-09-24:
### Distribution

Fedora Workstation 32 Kernel 5.8.10-200

### Support

I used the openrazer module for my new Deathadder v2 mouse. Thank you by the way. It worked great to allow me to change the led colors. Unfortunately as I began to get newer kernels (5.8.9, 5.8.10) things started to get unstable in my desktop session. Firstly during boot I am met a crash error saying the kernel is tainted or there was a kernel error. The "kernel tainted error" would specifically mention the  openrazer module as the cause. Shortly afterward, my monitor screens would go blank, as if the display port cable had been pulled out.

During kernel upgrades I see the openrazer module rebuilding successfully. With kernel 5.8.4, I get the kernel tainted error but the system is usable with no screen blanking. Removing the openrazer daemon, allows my machine to operate with no  problems on the newer kernels. 

I ssh'd into my machine when the error occurred and journalctl shows a 'page flip error'

If you need some specific logs I can reinstall the daemon and capture those for you.

z3ntu, 2020-09-26:
The kernel getting tainted is because an external module has been loaded which is nothing to worry about. You can read more about that at https://www.kernel.org/doc/html/latest/admin-guide/tainted-kernels.html

I mostly doubt that the driver causes e.g. the display parts to malfunction as we don't touch that at all. Our driver just gets used for the Razer usb devices on your system and shouldn't affect anything outside of that device.

But as you specifically mentioned
> Removing the openrazer daemon, allows my machine to operate with no problems on the newer kernels.

it looks like you tried running without openrazer already and that works..

malucious81, 2020-09-27:
Thanks for the response. I apologize I was mistaken. The problem did eventually show up again. I was finally able to find someone else experiencing the same issue. https://gitlab.freedesktop.org/drm/amd/-/issues/1297#note_636070

So amdgpu is to  blame. I will close this issue.

closed 2020-09-27 by malucious81