openrazer/openrazer #1266: Error with tertiary color on Kraken Ultimate headset

kevchu3, 2020-09-24:
### Distribution

Fedora 32, kernel 5.8.10-200.fc32.x86_64

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

### Reproduce Issue

I previously opened this issue at https://github.com/polychromatic/polychromatic/issues/287 but it was confirmed to be an issue with openrazer, and thus I am opening it here.

I am using the **Razer Kraken Ultimate headset** and pulled the latest commit 55c219f1d7f0e4e38d65259b0079b17436e37fdb from the master branch.  Steps to reproduce are as follows:

1. I set the Breath effect for my device using Single or Dual, and the headset colors work.

2. I then set the Triple color effect, and the headset colors also work and will cycle through all three colors.

3. I then go back and select either Single or Dual, which produces the stacktrace below. I've also noticed that any future selection of Single or Dual in the Breath effect, or even the single color from Static effect, will continue to produce this error. The only workaround is to unplug and replug in my device.

I've also noted when using polychromatic, that the Triple color effect is labeled `undefined`, which may be related.  I also noticed that the stacktrace is in `chroma_keyboard.py` but that isn't one of the devices I am using.

Stacktrace below:

```
Traceback (most recent call last):
  File "test-razer.py", line 12, in <module>
    device.fx.breath_dual(255,255,255, 128,128,128)  # Error happens
  File "/usr/lib/python3.8/site-packages/openrazer/client/fx.py", line 277, in breath_dual
    self._lighting_dbus.setBreathDual(red, green, blue, red2, green2, blue2)
  File "/usr/lib64/python3.8/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib64/python3.8/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib64/python3.8/site-packages/dbus/service.py", line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.8/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 479, in set_breath_dual_effect
    First byte is row, on firefly its always 1, on keyboard its 0-5
TypeError: 'tuple' object does not support item assignment
```

In a python3 interpreter:
```
import openrazer.client as rclient
device = rclient.DeviceManager().devices[0]  # First device
device.fx.breath_single(255,255,255)
device.fx.breath_dual(255,255,255, 128,128,128)
device.fx.breath_triple(255,255,255, 128,128,128, 255,255,255)
device.fx.breath_dual(255,255,255, 128,128,128)  # Error happens
```

z3ntu, 2020-09-26:
Caused by the (not yet merged) PR #1149 , so I'm closing the issue and noting it there.

closed 2020-09-26 by z3ntu

z3ntu, 2020-09-26:
I've fixed that issue in #1149 already btw