openrazer/openrazer #1267: Is it possible to help OpenRazer on Windows?

clementinise, 2020-09-24:
Hello!
I would like to contribute to OpenRazer, but it seems that I can't give any help since I'm on Windows, am I right?
Would really like to help get support for the Basilisk Ultimate and the Firefly V2 mousemat 😊

So is there any way we can still help other there even if we are on a Windows Distribution? Maybe a VM can be useful in this situation? (Sorry English is not my native language)

CalcProgrammer1, 2020-09-25:
OpenRazer is not officially supported on Windows.  I'm maintaining an unofficial Windows port here:

https://github.com/CalcProgrammer1/openrazer-win32

This project was originally created by user rsandoz a few years ago but was abandoned.  I revived it and got it to build against master openrazer with a minor patch applied.  It builds the OpenRazer kernel drivers into a Windows DLL.  The OpenRazer daemon components are not provided and the interface is modified as it's a DLL rather than a sysfs interface.

This DLL is used by my OpenRGB project on Windows.  OpenRGB supports a bunch of RGB devices from many vendors, OpenRazer is used to provide Razer support.

https://gitlab.com/CalcProgrammer1/OpenRGB

clementinise, 2020-09-25:
I'm using OpenRGB since a week but it doesn't support my mouse and mousepad and I have seen that we can get support for new device but the console commands are Linux and not Windows so I was wondering how is it possible to get new device supported even if we are on Windows, sorry if it wasn't clear btw 😅

CalcProgrammer1, 2020-09-25:
It is possible to write the changes in OpenRazer and then test them on Windows using my interface, but you'll have to rebuild both the DLL and OpenRGB (as OpenRGB has a file called OpenRazerDevices.h where all OpenRazer devices are listed and zones/LED maps are provided).

Have you checked to see if there are any open pull requests for your devices here on the OpenRazer github?

clementinise, 2020-09-25:
> Unfortunately, testing this way doesn't test the hardware probe functionality quite the same way as on Linux, so it's preferable that you test on a real Linux OS (and add OpenRazer daemon support) before you submit a pull request here.
Oh okay thx for the clear answer! 
> Have you checked to see if there are any open pull requests for your devices here on the OpenRazer github?
Yes there is a pull request for the [Basilisk Ultimate](https://github.com/openrazer/openrazer/issues/1215) and the [Firefly V2](https://github.com/openrazer/openrazer/issues/946) but I don't really understand at what state they are... I just want to help as best as I could since I really want to see all Razer devices supported that will be sooooo cool 😍

closed 2021-06-20 by github-actions[bot]