openrazer/openrazer #1269: Issue with Razer Naga Trinity, BlackWidow Chroma V2 and Kraken Ultimate

RPavlik1, 2020-09-26:
### Distribution

Razer Naga Trinity
Razer BlackWidow Chroma V2
Razer Kraken Ultimate
### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The DPI Settings do not stick on my Naga Trinity when the computer is turned off for a long period of time and turned back on. 
The Lighting on the Black Widow ChromaV2 settings don't save either. 
Kraken Ultimate is not detected on RazerGenie.

Computer has been restarted for the Mouse and Keyboard issue and it doesn't appear to be an issue. It only seems to happen if I turn off my desktop for the night and turn it back on in the morning.

The Kraken Ultimate headset works for talking and hearing stuff, however I do not have any settings for lights or mic monitoring, if that is an option with this. 

I am setup in plugdev group, any ideas on what I am doing wrong.

z3ntu, 2020-09-26:
Can you give the output of `lsusb | grep 1532` please?

RPavlik1, 2020-09-26:
> Can you give the output of `lsusb | grep 1532` please?

Here is the output you requested.

Bus 003 Device 004: ID 1532:0221 Razer USA, Ltd Razer BlackWidow Chroma V2
Bus 003 Device 003: ID 1532:0527 Razer USA, Ltd Razer Kraken Ultimate
Bus 003 Device 002: ID 1532:0067 Razer USA, Ltd Razer Naga Trinity

z3ntu, 2020-09-26:
I'm guessing you're running the latest v2.8.0 release where the Kraken Ultimate isn't supported yet - it will be in the next release. We will have some code for remembering effects/settings on the openrazer side soon with #1149 but generally the DPI should stay set across power "outages"..

RPavlik1, 2020-09-26:
> I'm guessing you're running the latest v2.8.0 release where the Kraken Ultimate isn't supported yet - it will be in the next release. We will have some code for remembering effects/settings on the openrazer side soon with #1149 but generally the DPI should stay set across power "outages"..

Yeah, I am running v2.8.0
The DPI setting seems a bit inconsistent. I just powered on today and everything was fine, where it didn't save yesterday morning during bootup, so things are a little weird.

Thanks for the prompt response man. I really appreciate you getting back to me so fast. I will wait for the next update for Ultimate's Support. Thanks for the help!

z3ntu, 2020-09-26:
No problem. And generally the DPI is saved by the firmware so I don't think we can do more than tell the device to use that DPI. Do you have any further questions?

RPavlik1, 2020-09-26:
> No problem. And generally the DPI is saved by the firmware so I don't think we can do more than tell the device to use that DPI. Do you have any further questions?

Nope, I don't have any other questions. If you have any advice on how to troubleshoot the DPI thing, I would greatly appreciate it. I just moved to Linux Mint like a week or so ago, so all this is very new to me lol.

z3ntu, 2020-09-26:
Do you know what DPI it resets to after the night and if it's always the same every time

RPavlik1, 2020-09-26:
> Do you know what DPI it resets to after the night and if it's the same every time?

The DPI resets to 1800 after each night, or when it occurs at least. I changed it back to 800.

RPavlik1, 2020-10-02:
I just wanted to update this issue. I think I found the cause to the problem. 

When I was not using my desktop and razer mouse,headset and keyboard on my desktop. I would take them out and pop them into a docking station to use for my work laptop. Its only when doing this and putting them back into the desktop that it would have to have the DPI and lights reset. 

Not sure if this is a bug or not, just wanted to let you guys know I figured it out. Should I close the issue or is this not intended?

z3ntu, 2020-10-03:
It's interesting behavior for sure but I don't think there's anything we can do to prevent this from happening.

closed 2020-10-03 by z3ntu