openrazer/openrazer #1273: razer deathadder elite not detected by RazerGenie but is detected by my system as razer deathadder elite

ProficientPug, 2020-09-28:


ProficientPug, 2020-09-28:
also one more thing: this app is using daemon version 2.8.0(idk about version numbers so i think it might be outdated software

lah7, 2020-09-29:
At this time of writing, OpenRazer 2.8.0 is the latest and RazerGenie is 0.9.0.

Most likely the driver isn't installed correctly, so none of the applications will work. Please take a look over the [troubleshooting page](https://github.com/openrazer/openrazer/wiki/Troubleshooting#kernel-modules-building). What is the output when you run this?

    sudo dkms install openrazer-driver/2.8.0

Also, check the daemon to be sure there are no errors like user is not part of the `plugdev` group:

    openrazer-daemon -Fv

closed 2021-06-20 by github-actions[bot]