openrazer/openrazer #1274: Razer Trinity Naga

JavelinHeadQueer, 2020-09-29:
Color option is stuck on Static despite Windows being able to set it to other colours. Is this something I can solve by rebooting or is it a deeper issue with the software?

lah7, 2020-09-29:
This device  (`1532:0067`) supports the static effect, so it is possible to change to another RGB colour.

Which application are you using or how are you changing the effect?

Try interfacing with the OpenRazer Python library directly via a Terminal:

    python3 -c "import openrazer.client as rclient; rclient.DeviceManager().devices[0].fx.static(255,0,0)"

:point_up:  This will set the colour to red (255,0,0), assuming it is the first Razer device.

JavelinHeadQueer, 2020-09-29:
I'm using the Polychromatic Controller. The mouse in Windows supports other effects than static, and if I don't set a static colour for it, it breathes in different colours, yet more settings than static are unavailable.

lah7, 2020-09-29:
OpenRazer interfaces with the device's firmware "hardware" effects, which is what Polychromatic (and other applications) will expose too. There should be options to change the colour under the drop down.

This particular device is only known to support static, DPI, poll rate and brightness:

https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L936-L952

It's possible that this device heavily relies on the Synapse 3 driver to perform "software" effects, which OpenRazer exposes as a matrix API for scripts to control each LED.

In future, Polychromatic aims to make it possible to create software effects. Until then, there is a bug that might cause the matrix draw call to fail for this mouse: https://github.com/openrazer/openrazer/issues/1252

closed 2021-06-20 by github-actions[bot]