openrazer/openrazer #1276: Razer Naga Trinity Not supported

RPavlik1, 2020-10-02:
### Distribution

Razer Naga Trinity
Linux 
Bus 003 Device 002: ID 1532:0067 Razer USA, Ltd Razer Naga Trinity

NAME="Linux Mint"
VERSION="20 (Ulyana)"
ID=linuxmint
ID_LIKE=ubuntu

### Support

- [ X ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Cannot use the custom editor and clicking in on the mouse wheel does not let me scroll. 
I can use the mousewheel to scroll up and down though. That works just fine.

[image]

z3ntu, 2020-10-03:
Hi, the "custom editor" in RazerGenie for mice is tracked in https://github.com/z3ntu/RazerGenie/issues/10 but it's not needed for anything except for configuring very custom lighting patterns.

> and clicking in on the mouse wheel does not let me scroll.

That's a general Linux-thing, middle click is pasting the secondary clipboard iirc. It seems that with GNOME you can disable that feature; or e.g. in web browsers you can install extensions to enable the middle click scrolling.

closed 2020-10-03 by z3ntu

RPavlik1, 2020-10-04:
Thanks for letting me know, still real new to linux in general. =)