openrazer/openrazer #1278: Support for Deathadder 1800 variant

philectro, 2020-10-03:
### Device Info

* Name: DeathAdder Mouse
* Product Number: i can't tell as the sticker were wearing and i removed it
* Model Number: see above, if i find the box i post more informations about it

### Device Information
```
Bus 005 Device 003: ID 1532:0007 Razer USA, Ltd DeathAdder Mouse
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0007 DeathAdder Mouse
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 DeathAdder
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      73
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx 1 root root   9 Okt  4  2020 usb-Razer_DeathAdder-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Okt  4  2020 usb-Razer_DeathAdder-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
I will do it later!

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
I will do it later too :)

achiya-SX, 2023-08-08:
Hi guys, anyway this issue got fixed?

z3ntu, 2023-08-08:
Normally I would point you to https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md but since this device is super old, it probably needs a bunch of custom code like Orochi 2011 and some old DeathAdder (see razermouse_driver.c)