openrazer/openrazer #1279: Support for BlackWidow V3 Pro

apachano, 2020-10-04:
### Device Info

* Name: BlackWidow V3 Pro
* Product Number: RZ03-03530200-R3U1
* Model Number: RZ703-0353

### Device Information

<details>
  <summary> lsusb -d 1532: -v </summary>

  ```
Bus 001 Device 003: ID 1532:025c Razer USA, Ltd Razer BlackWidow V3 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x025c 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
  ```
</details>
    
<details>
  <summary>ls -lah /dev/input/by-id/</summary>

  ```
total 0
drwxr-xr-x 2 root root 240 Oct  3 20:34 .
drwxr-xr-x 4 root root 800 Oct  3 20:34 ..
lrwxrwxrwx 1 root root   9 Oct  1 17:28 usb-Logitech_USB_Receiver-if02-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Oct  1 17:28 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Oct  1 17:28 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-if01 -> ../event27
lrwxrwxrwx 1 root root  10 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-if03 -> ../event29
lrwxrwxrwx 1 root root  10 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if01-event-kbd -> ../event24
lrwxrwxrwx 1 root root  10 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if02-event-mouse -> ../event28
lrwxrwxrwx 1 root root   9 Oct  3 20:34 usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if02-mouse -> ../mouse4
lrwxrwxrwx 1 root root   9 Oct  1 17:28 usb-SteelSeries_Arctis_Pro_Wireless-event-if05 -> ../event7
  ```
</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
(Coming soon, need to plug this into a friends computer)

slayer1992, 2021-04-12:
I have recently purchased the Blackwidow V3 Pro and would like support. Is there any information I can help acquire?

billygilbrech, 2021-04-22:
I recently bought this board as well, I can provide anything you need. Please get back to me. :)

apachano, 2021-04-22:
@slayer1992 @billygilbrech I ended up returning my KB so I can't do this, but this is the next step that is needed for this board https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

slayer1992, 2021-04-23:
@apachano thanks for the info! I will try and get set up for gathering packet data. It is a crazy busy week unfortunately...

GeoDoX, 2021-04-23:
The packet data is already collected iirc, just need a dev willing to put
it all together.

j-maynard, 2021-09-22:
I've been looking at the BlackWidow Pro V3 and The Naga Pro...  I should probably draw peoples attention to the fact that if you have your Hyperspeed Dongle used for Both Mouse and Keyboard the the mouse in my case the Naga Pro is a reported to XInput as Razer BlackWidow V3 Pro Mouse... See the screenshot:

[image]

My guess is that if you do the same with the Razer HyperSpeed Dongle which comes with the Naga the device will report as a Razer Naga Pro Keyboard.  I've not tested that yet... I'll cross post this information to the the Naga Pro Support issue as well.

GeoDoX, 2021-09-28:
Trying to work on support for this, however I'm not getting an usb hid interface over bluetooth for this device.

jasclarke24, 2022-04-28:
What is the status of this commit? It is not working with version 0.7.3 on Ubuntu 22.04 LTS.

ChappIO, 2022-07-01:
I just tested with my Pro v3 and when I connect it wireless (1532:025C) it is an unrecognized device. But through USB (1532:025A) it works fine.

Environment info:

```
$ openrazer-daemon --version
openrazer-daemon 3.3.0

$ hostnamectl
 Static hostname: blade
       Icon name: computer-laptop
         Chassis: laptop
      Machine ID: b5608ea86df4fda8ad12970c62bd3fac
         Boot ID: 4218dc61175a4cf99329dc28cd53cbc0
Operating System: Pop!_OS 22.04 LTS                              
          Kernel: Linux 5.17.15-76051715-generic
    Architecture: x86-64
 Hardware Vendor: Razer
  Hardware Model: Blade 15 Advanced Model _Early 2021_ - RZ09-036

$ polychromatic-cli --version
Polychromatic 0.7.3
Save Data: 8
Python: 3.10.4
```

atombombzero, 2022-08-09:
TLDR: Has  this been sufficiently tested through 1622 and 1623?

1532:025A - Wired (works just fine)
1532:025C - 2.4Ghz dongle (does not work)
---------------------------------------
root@arm-ubuntu2004:~# openrazer-daemon --version
openrazer-daemon 3.4.0
root@arm-ubuntu2004:~# polychromatic-cli --version
Polychromatic 0.7.3
Save Data: 8
Python: 3.8.10

root@arm-ubuntu2004:~# hostnamectl
   Static hostname: arm-ubuntu2004
   Pretty hostname: arm-ubuntu20_04
         Icon name: computer-laptop
           Chassis: laptop
        Machine ID: c26f23b44f20474d8f6a35fe411b0c4c
           Boot ID: 157201f9fdda4b47a2819c89952a069b
  Operating System: Ubuntu 20.04.4 LTS
            Kernel: Linux 5.14.0-1047-oem
      Architecture: x86-64

root@arm-ubuntu2004:~# lsusb -d 1532: -v
Bus 003 Device 018: ID 1532:025a Razer USA, Ltd Razer BlackWidow V3 Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x025a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V3 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
[... 262 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

ChappIO, 2022-08-10:
@atombombzero probably not as #1623 is still open

closed 2024-07-03 by z3ntu