openrazer/openrazer #1280: Support for BlackShark V2 Pro

mohammed-n-shamsi, 2020-10-07:
### Device Info

* Name: Razer BlackShark V2 Pro
* Product Number: RZ04-03230100-R3U1 
* Model Number: RZ04-0323

### Device Information
```
    lsusb -d 1532: -v

[lsusb_output.txt](https://github.com/openrazer/openrazer/files/5342617/lsusb_output.txt)

```

```
Please post the output of
    ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root   9 Oct  6 19:05 usb-Razer_Razer_BlackShark_V2_Pro-event-if03 -> ../event4

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
Will add soon.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Will add soon.

marcemira, 2020-11-16:
Any updates here?

cyferc, 2020-11-16:
> Any updates here? Is this the same as #1254 but for Pro version @cyferc?

Unlikely the same. Pro is Bluetooth enabled and has a different product ID.

mohammed-n-shamsi, 2020-11-16:
Sorry, forgot I even made this, I will sit down this upcoming weekend to learn and capture the packets.

Benzer1406, 2021-01-20:
Any updates? I'm considering buying the V2 Pro but am not sure if it is compatible with Linux. Anybody can give me a hint? (I use Kubuntu with Ubuntu 20.04)

ceiphr, 2021-01-20:
@Benzer1406 Just got mine. Everything works except the microphone on Ubuntu 20.10. I'm currently using my system's internal microphone as a substitute.

Hopefully, we can see progress on this issue, so the microphone can work in the future.

Benzer1406, 2021-01-20:
@ceiphr  Thanks for the info. What distribution do you use? Have you OpenRazer installed and tried with that? @mohammed-n-shamsi do you think there might be an update soon?

ceiphr, 2021-01-20:
@Benzer1406 Standard Ubuntu 20.10 with GNOME on Wayland. OpenRazer and Polychromatic installed.

Benzer1406, 2021-01-21:
@ceiphr https://github.com/openrazer/openrazer/issues/1254#issuecomment-746855753

Here somebody got it to work by changing some pulse audio settings. Could be helpful to look into that...

The mic works fine when you connect the cable, right?

ceiphr, 2021-01-21:
@Benzer1406 Thanks for that. 

> @ceiphr #1254 (comment)
>
> Here somebody got it to work by changing some pulse audio settings. Could be helpful to look into that...

I made a comment in that issue just now to see what pulseaudio settings were used.

> The mic works fine when you connect the cable, right?

I just tried with the cable, still doesn't work. But, it's important to note that my AUX port is a bit janky, the output audio sounds a bit low/muffled too.

Benzer1406, 2021-01-21:
@ceiphr oh wow, that scares me. I just ordered my v2 Pro today. I read in 2 articles though that it works with Linux. What is weird is that with a normal audio jack cable the mic doesn't work. I suppose you used the one with three rings?!

I use Kubuntu with Ubuntu 20.04. I hope it'll work at least with the cable. 
Do you have a chance to check if it's your audio jack port or really the headset? (Maybe with a different headset or a different computer. 

Not sure what else could be done...so sad as it's the premium model and quite expensive.

ceiphr, 2021-01-21:
@Benzer1406 I'll check tonight with another AUX-based headset.

> EDIT: Just saw the response in the other issue. Please let me know if it worked for you. I'm not too interested in the fancy sound controls, rather being able to use the headset and mic wirelessly.

I will also test out those suggested tweaks tonight. I'll let you know how it goes.

Benzer1406, 2021-01-24:
@ceiphr Thanks for testing. I still have hope that the wireless will work as I read this thread (https://www.reddit.com/r/linux_gaming/comments/jyip39/razer_blackshark_v2_pro_poposubuntu_support/)

In the worst case  I will use it with the cable, I ordered a 10ft cable which will give me enough liberty even though I'd obviously prefer using it wirelessly. I'm not sure what the packet capture is?

Do you think there is a chance of getting this project to officially support this model?

Thanks in any case for your help and effort.

ceiphr, 2021-01-24:
@Benzer1406 I'm happy to hear other people could confirm it works!

> Do you think there is a chance of getting this project to officially support this model?

I hope. I'm going to learn how to capture packets, so the devs can having something to work with. Although, I'm unsure when I can have that done.

Benzer1406, 2021-02-02:
Hi @ceiphr ,
I just got my Blackshark V2 Pro. As a reminder, I run Kubuntu 20.04. I plugged in the USB from Razer. It took 2 seconds and the system recognized speakers and a mic. I checked both and they work perfectly. The speakers sound really good and I checked the mic through a Zoom Call and it sounded fine, very close to what I listened to on YouTube.

I checked the cable and it worked perfectly as well but with a little bit less quality in both. 

Super happy!!

pozsa, 2021-12-07:
The mic does not work for me in wireless mode. Can anyone point to an article or explain how to do `pcap` for the headset?

z3ntu, 2021-12-07:
FYI OpenRazer support will change nothing about the audio bits, OpenRazer only controls lights on the device.

ede1998, 2022-03-26:
> FYI OpenRazer support will change nothing about the audio bits, OpenRazer only controls lights on the device.

I own the BlackShark V2 Pro and as far as I can tell there is no RGB on it. Just a status LED. What I would be interested in though is being able to see the battery level with OpenRazer. Is that a use case that OpenRazer supports? If so, I can try to make it work for this headset.

z3ntu, 2022-03-26:
@ede1998 Yes, battery level (if through the normal API Razer uses) is in scope of openrazer.

ede1998, 2022-04-09:
I followed the [guide](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) and made a few [packet captures](https://cloud.erik-hennig.me/s/HAdQeRNnmzxwtBZ)

Request charging state (device not charging)*:
```
0000   40 72 d5 23 af 93 ff ff 53 02 00 03 03 00 00 00
0010   61 af 40 62 00 00 00 00 dc 77 02 00 8d ff ff ff
0020   40 00 00 00 40 00 00 00 21 09 ff 03 03 00 40 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   ff 0a 00 fd 04 12 f1 02 05 00 00 00 00 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Now I'm a bit taken aback by how the packet looks. In the guide, it says the packet (usually) starts with `00ff` but my data fragment starts with `ff`. Also it does not seem to contain the 3 reserved bytes `000000` indicated in the guide.

@z3ntu Do you have any suggestions/pointers for me? Am I reading the packet wrong? Or is it a completely different layout/protocol for this headset?

\* at least I think that's what the packet capture contains. This sequence of packages was triggered when I selected the device in Synapse and it refreshed the charging state when I selected it.

z3ntu, 2022-04-10:
We do see these packets in the pcaps:
```
Status Trans Packet Proto DataSize Class CMD Args
ff     0a    00fd   04    12       f1    02  050000000000000000000000000000000000
ff     0a    00fd   04    12       f1    02  410000000000000000000000000000000000
ff     0a    00ff   04    12       f1    05  410084030000000000000000000000000000
ff     0a    00ff   04    12       f1    05  4101100e0000000000000000000000000000
ff     0a    00ff   04    12       f1    05  410184030000000000000000000000000000
```
I've never seen those before so seems like they're using new functions in this device

For f1 05 the second bit is whether power saving is on.

3rd and 4th argument seems to depend on the power saving time:
* 0x84 0x03 = 15 minutes (little endian, so 0x0384 is 900 seconds = 15 minutes)
* 0x10 0x0e = 60 minutes (little endian, so 0x0e10 is 3600 seconds = 60 minutes)

---

Not really sure what f1 02 does, in the "is charging" pcaps the arguments are always the 05000000... one. And in the other pcaps it's always the 41000000... one

ede1998, 2022-04-18:
Thanks for the quick reply.
I made a few [more packet captures](https://cloud.erik-hennig.me/s/MzgDWsqsmNPpToN) but they don't really seem to fit the scheme you were describing in your comment :/

```
direction  Status	Trans	Packet	Proto	DataSize	Class	CMD	Args not sure if this is right...
host > dev FF		0A	00FD	04	00		00	05	8160 C001 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FD	04	00		00	05	8160 C010 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FD	04	00		00	05	81AA C601 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FF	04	00		00	02	8400 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FF	04	00		00	0C	002E C400 7F00 0060 C010 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FF	04	00		00	0C	002E C410 7F00 0060 C001 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FF	04	0E		F1	05	0104 00C0 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
host > dev FF		0A	00FF	04	0E		F1	05	0104 00D7 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000

dev > host FF		01	00FE	00	04		0F	05	8160 C000 4E32 3134 3344 3133 3030 3033 3634 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
dev > host FF		0C	05FE	00	04		0F	05	8160 C000 0EA8 5034 3344 3133 3030 3033 3634 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
dev > host FF		0C	05FE	00	04		0F	05	8160 C000 4E32 3134 3344 3133 3030 3033 3634 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
dev > host FF		0C	05FE	00	04		0F	05	81AA C601 4E32 3134 3344 3133 3030 3033 3634 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
dev > host FF		1B	05FE	00	04		0F	14	8160 C048 4E32 3134 3344 3133 3030 3033 3634 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
```
The `dev > host` packets are replies to the previous requests (order does not match in my code block). The `4E32 3134 3344 3133 3030 3033 3634` is the device's serial number. I also ran the [firmware update tool](https://mysupport.razer.com/app/answers/detail/a_id/4727) and got packet captures that show the firmware version (though only for the transceiver not the headset itself, the headset has to be connected via USB directly, not wirelessly to read the firmware afaict).

I started working on implementing the driver (though I'm really unsure at the moment how correct packets should look like...). Do you have any suggestion how I can test the driver without actually sending the generated packets to my physical device (so I do not accidently brick it). I tried googling and found stuff like GadgetFS but I'm not really sure if that's what I need. I'd simply like to emulate the headset as a dummy usb device that does not react at all when I request device specific functions (or maybe lets me handcraft replies).

nazarred, 2022-07-20:
> Hi @ceiphr , I just got my Blackshark V2 Pro. As a reminder, I run Kubuntu 20.04. I plugged in the USB from Razer. It took 2 seconds and the system recognized speakers and a mic. I checked both and [...]

It also works for me (mic and sound) using a USB from Razer. I'm on Ubuntu 20.04.4 LTS (5.15.0-41-generic).

Tattersson, 2023-01-02:
Any updates on getting Support for the headset?

Tattersson, 2023-01-02:
Any updates on getting the support? 
Headset works fine, but would love to be able to configure it more.

Richard-Payne, 2023-01-23:
> FYI OpenRazer support will change nothing about the audio bits, OpenRazer only controls lights on the device.

The mic works fine wirelessly for me.  The thing I'm really missing is the sound shaping that the Windows driver software does.  Is this something that OpenRazer is looking to address?

z3ntu, 2023-02-11:
@Richard-Payne Depends on if it just means sending some command to the device to do that. If it's just a software-side feature then it's out-of-scope of OpenRazer and you can use some vendor-agnostic software that might be able to do this.

Max-feded, 2023-02-13:
> Any updates on getting the support? Headset works fine, but would love to be able to configure it more.

me, too

Tortillas-IT, 2023-04-05:
would be fine to see at least the battery level of the headset, maybe this could extended to all/most wireless/bluetooth razer's headset

fiercebrake, 2023-06-27:
looking forward to have this supported!

thanks!

MonadMania, 2024-06-05:
How do I get the USB features to work on Fedora? Nothing happens when I plug it in.

cipopescu-dev, 2024-07-09:
Hello! 2023 version here, on Debian12 (via usb wireless): Both mic and speakers works well for 10-15 minutes. After this period, the mic stops working without any message about it. (Speakers are ok, only the mic has this problem)

Tortillas-IT, 2024-07-10:
> Hello! 2023 version here, on Debian12 (via usb wireless): Both mic and speakers works well for 10-15 minutes. After this period, the mic stops working without any message about it. (Speakers are ok, only the mic has this problem)

that's your distro issue, openrazer doesn't support headset devices

SourceE1, 2024-09-17:
Just being able to check battery level for this headset would be such a massive addition to me, so while I am a complete novice to such development  I would like to understand if there is anything I can do to help try get some data to make this feature request possible?

tcr27, 2025-01-27:
I was able to filter the relevant packets and created a python script to get the battery level.
[Razer BlackShark Batter Info](https://github.com/tcr27/razer-blackshark-battery-info). 

I also tried to add it to the driver/daemon, but my knowledge of C and Python is currently too basic.

Tortillas-IT, 2025-01-28:
> I was able to filter the relevant packets and created a python script to get the battery level. [Razer BlackShark Batter Info](https://github.com/tcr27/razer-blackshark-battery-info).
> 
> I also tried to add it to the driver/daemon, but my knowledge of C and Python is currently too basic.

bump!

tcr27, 2025-02-14:
I investigated a bit more and was able to extract some commands and map them to a capture i took.

[blackshark_mapping.tar.gz](https://github.com/user-attachments/files/18803152/blackshark_mapping.tar.gz)

rcastelar, 2026-11-12:
I would love for this to be merged