openrazer/openrazer #1283: Keyboard and mouse not recognized by BIOS immediately during POST

marcus-cr, 2020-10-10:
Settings seem to take affect outside of Linux (during POST and in BIOS). 

Also my PC now takes 30+ seconds just to recognize keyboard and mouse. Only after this time, when both light up to my RGB color settings, am I able to enter BIOS setup or boot any OS. In the past it was recognized immediately, albeit with default colors. 

Tried unplugging and replugging in devices, resetting the onboard profiles on a Windows machine, loaded default BIOS settings. 

Bit alarmed that this is persistent and affecting boot times. 

Devices are:
Basilisk
- Product number RZ01-02330100
- Model number RZ01-0233
BlackWidow Elite
- Product number RZ03-02621900-R3U1
- Model number RZ03-0262

marcus-cr, 2020-10-11:
Linux seems to have messed with my powered USB hub. Changed the port and the noted issues are gone, even though none of my Razer devices were plugged into the hub.

Only issue that remains is the Basilisk seems to persist with the color I set.

z3ntu, 2020-10-11:
> Only issue that remains is the Basilisk seems to persist with the color I set.

Can you please elaborate on that problem?

marcus-cr, 2020-10-12:
> > Only issue that remains is the Basilisk seems to persist with the color I set.
> 
> 
> 
> Can you please elaborate on that problem? 

I removed OpenRazer but the mouse RGB color persists.

Apologize for the disorganization. The initial issue probably should've been closed, since this is a separate one.

z3ntu, 2020-10-13:
Please explain the problem, I don't know what you're talking about.

marcus-cr, 2020-10-13:
This is separate issue. Closing this, opened new ticket (#1290).

closed 2020-10-13 by marcus-cr