openrazer/openrazer #1285: Is Cynosa Lite supported?

matteocasonato, 2020-10-12:
### Distribution

Linux Ubuntu 20.04

### Problem

I've followed the few steps needed to install Drivers and Polychromatic Controller. 
I've executed the command `sudo gpasswd -a $USER plugdev`

I think I made everything well, then I saw in the main page [https://openrazer.github.io/](url) Cynosa Lite is not supported. 
By the way, Cynosa Chroma is supported, so I thought the problem is caused by me. 

Am I wrong? Maybe Cynosa Lite is completely incompatible?

I've tried on Windows, and the Keyboard works fine.

lah7, 2020-10-12:
Cynosa Lite (1532:023F) is supported, but there hasn't been a new OpenRazer release yet (v2.9.0)

In the meantime, you can add the "daily" repository and install those packages. :+1:

    sudo add-apt-repository ppa:openrazer/daily
    sudo apt dist-upgrade

matteocasonato, 2020-10-12:
Okay thanks a lot! I'll wait for it then.

closed 2020-10-13 by z3ntu