openrazer/openrazer #1286: Support for Kraken Kitty Edition nonfunctional

NateDogg1232, 2020-10-12:
### Distribution

Arch Linux
Kernel ver: 5.8.14-arch1-1
Openrazer ver: Master branch commit 2399787
Installed using openrazer-meta-git

### Support

The new support for the Razer Kraken Kitty Edition is not working. Upon starting the daemon (Both using systemd and just using `openazer -Fv`, the daemon will crash out with this error message:

```
KeyError: "Can't register the object-path handler for '/org/razer/device/HN1947D11900659': there is already a handler"
```

If needed, I can send the entire log that I have. I have verified using `dmesg` that the driver is working and running.

It should be noted that my log files do look like this:
```
2020-10-12 10:23:46 | razer                          | INFO     | Found device.0: 0003:1532:0F19.0005
2020-10-12 10:23:46 | razer.device0                  | INFO     | Initialising device.0 RazerKrakenKittyEdition
2020-10-12 10:23:46 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
[...]
2020-10-12 10:23:46 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2020-10-12 10:23:46 | razer                          | INFO     | Found device.1: 0003:1532:0F19.0006
2020-10-12 10:23:46 | razer.device1                  | INFO     | Initialising device.1 RazerKrakenKittyEdition
```

I notice that there are two "RazerKrakenKittyEdition" devices being initialized, but I should also note that I do only have one single  headset plugged in, verfied through `lsusb` as well (There IS some sort of USB hub that this has as well, but I don't know if that is important)

MegaShinySnivy, 2020-10-13:
Can confirm, I'm getting the exact same errors Nate is getting.

tbaumann, 2020-10-13:
Strange I just today got the same error.

I am using a recently merged fork to support the RazerKrakenKittyEdition

```
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2020-10-13 13:03:47 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2020-10-13 13:03:47 | razer                          | INFO     | Found device.1: 0003:1532:0F19.000D
2020-10-13 13:03:47 | razer.device1                  | INFO     | Initialising device.1 RazerKrakenKittyEdition
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 121, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 107, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 401, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 489, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 452, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 84, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/service.py", line 87, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 489, in __init__
    self.add_to_connection(conn, object_path)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 578, in add_to_connection
    connection._register_object_path(path, self._message_cb,
KeyError: "Can't register the object-path handler for '/org/razer/device/HN1943D12008311': there is already a handler"

Stopping daemon.
```

I can't see any other instance of the daemon running.

tbaumann, 2020-10-13:
Reproduced with build from current master.

23997877467790800188c5371159f16d24b7b79c

NateDogg1232, 2020-10-13:
I'm sorry but I don't quite understand what you mean by "Putting the case statement onto one of the other options"

Like, make it return some other `expected_protocol`?

z3ntu, 2020-10-13:
Try that if that helps, it helped on other devices. I don't have the device so it's difficult to figure that out for me.

NateDogg1232, 2020-10-13:
So I tried with both `USB_INTERFACE_PROTOCOL_MOUSE` and `KEYBOARD` and both didn't work.

z3ntu, 2020-10-13:
CC @999eagle

tbaumann, 2020-10-13:
The odd thing is that this had worked well for many days. I wonder what outside factors changed.

NateDogg1232, 2020-10-13:
Well, I'm going to try to revert to the commit that the support was added and see if there has been any regression since

NateDogg1232, 2020-10-13:
No matter how hard I try, I cannot get the dkms modules for the commit the support was added running. I can `modprobe` them in, but when I do I don't get a single message in `dmesg`

NateDogg1232, 2020-10-13:
I take it back. Of course the moment I write that I think of something.

So, yeah it hasn't worked since the original commit for me, as I just tried that commit. I wish I knew a bit more about how the drivers worked, but I really don't.

z3ntu, 2020-10-13:
@NateDogg1232 Have you tried commit c1160714087d1d11a59abbce36e0706509fd4291 ?

NateDogg1232, 2020-10-13:
Just did right now and yes it is still the case then

MegaShinySnivy, 2020-10-14:
I can also confirm this is still an issue. I noticed--and this could be unrelated, I don't know--but in these lines https://github.com/openrazer/openrazer/blob/f3f06ffedafb86898a0c5b00da328fe1e30a655f/driver/razeraccessory_driver.c#L329-L332
The value `report.transaction_id.id` does not match the others, as it is set to `0x1F` and not `0x3F` like the other devices.

999eagle, 2020-10-14:
I've tried to debug this as well now. First of all, the kernel driver reports two devices added whenever I plug in the headset:

```
[104499.913809] usb 1-3.1: new high-speed USB device number 24 using xhci_hcd
[104500.007058] usb 1-3.1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice= 5.11
[104500.007060] usb 1-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[104500.007061] usb 1-3.1: Product: USB2.0 Hub
[104500.007061] usb 1-3.1: Manufacturer: GenesysLogic
[104500.071580] hub 1-3.1:1.0: USB hub found
[104500.072183] hub 1-3.1:1.0: 4 ports detected
[104500.414807] usb 1-3.1.2: new full-speed USB device number 25 using xhci_hcd
[104500.512322] usb 1-3.1.2: New USB device found, idVendor=1532, idProduct=0f19, bcdDevice= 2.00
[104500.512324] usb 1-3.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[104500.512325] usb 1-3.1.2: Product: Razer Kraken Kitty Chroma Control
[104500.512326] usb 1-3.1.2: Manufacturer: Razer
[104500.589504] input: Razer Razer Kraken Kitty Chroma Control as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2:1.0/0003:1532:0F19.0024/input/input68
[104500.641860] razeraccessory 0003:1532:0F19.0024: input,hidraw2: USB HID v1.11 Device [Razer Razer Kraken Kitty Chroma Control] on usb-0000:07:00.1-3.1.2/input0
[104500.645387] razeraccessory 0003:1532:0F19.0025: No inputs registered, leaving
[104500.645409] razeraccessory 0003:1532:0F19.0025: hidraw13: USB HID v2.00 Device [Razer Razer Kraken Kitty Chroma Control] on usb-0000:07:00.1-3.1.2/input1
[104501.621805] usb 1-3.1.3: new full-speed USB device number 26 using xhci_hcd
[104501.932340] usb 1-3.1.3: New USB device found, idVendor=1532, idProduct=0521, bcdDevice= 0.70
[104501.932342] usb 1-3.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[104501.932343] usb 1-3.1.3: Product: Razer Kraken Kitty Edition
[104501.932344] usb 1-3.1.3: Manufacturer: Razer
[104501.932345] usb 1-3.1.3: SerialNumber: 00000000
[104502.421608] input: Razer Razer Kraken Kitty Edition Consumer Control as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.3/0003:1532:0521.0026/input/input69
[104502.473845] input: Razer Razer Kraken Kitty Edition as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.3/0003:1532:0521.0026/input/input70
[104502.473918] hid-generic 0003:1532:0521.0026: input,hiddev0,hidraw14: USB HID v1.11 Device [Razer Razer Kraken Kitty Edition] on usb-0000:07:00.1-3.1.3/input3
```

Somehow device numbers `0003:1532:0F19.0024` and `0003:1532:0F19.0025` are added. Both of these devices are enumerated in `/sys/bus/hid/devices/[id]/` and have all the device files the kernel driver adds. Both can control the headset.

During my testing, the daemon worked perfectly fine because I always started the daemon before plugging in the headset. Logs for this look like this:

```
2020-10-14 08:51:03 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2:1.0/0003:1532:0F19.0027
2020-10-14 08:51:03 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2:1.1/0003:1532:0F19.0028
2020-10-14 08:51:03 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2:1.1/0003:1532:0F19.0028
2020-10-14 08:51:03 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2:1.0/0003:1532:0F19.0027
2020-10-14 08:51:04 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.3/0003:1532:0521.0029
2020-10-14 08:51:04 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.1/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.3/0003:1532:0521.0029
2020-10-14 08:51:05 | razer                          | INFO     | Found valid device.0: 0003:1532:0F19.0028
2020-10-14 08:51:05 | razer.device0                  | INFO     | Initialising device.0 RazerKrakenKittyEdition
2020-10-14 08:51:05 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
[snip]
2020-10-14 08:51:05 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2020-10-14 08:51:05 | razer                          | DEBUG    | Emitted Device Added Signal
```

If the daemon is started with the headset already plugged in, it seems to try adding both devices:

```
2020-10-14 08:52:41 | razer                          | INFO     | Found device.0: 0003:1532:0F19.0027
2020-10-14 08:52:41 | razer.device0                  | INFO     | Initialising device.0 RazerKrakenKittyEdition
2020-10-14 08:52:41 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
[snip]
2020-10-14 08:52:41 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
[... cut ...]
  File "/usr/lib/python3.8/site-packages/dbus/service.py", line 578, in add_to_connection
    connection._register_object_path(path, self._message_cb,
KeyError: "Can't register the object-path handler for '/org/razer/device/HN1943D11901317': there is already a handler"

Stopping daemon.
```

I'm not sure how this should be handled to be honest. For now this is just my analysis of why this happens, but I don't know how we can filter this (preferably in the kernel driver) and only add the device once.

MegaShinySnivy, 2020-10-17:
I think I figured out why the RKKE headphones show up as two devices. I was booted into Windows at the moment, messing around with a VM when I noticed that my headphones showed up as "Razer Kraken Kitty Edition" and "Razer Kraken Kitty Edition Chroma Controler" for VM-passthrough. Have they somehow stuck two USB devices in one? If so, that could explain our issues. I'll provide screenshots if they'd help.

999eagle, 2020-10-17:
Actually it's three USB devices stuck in one: a USB hub, the Razer Kraken Kitty Edition (which is the audio device) and the Razer Kraken Kitty Edition Chroma Controller (which is for controlling the lighting). All three devices use different VID/PIDs though, so that's not the problem. The problem is that the Chroma Controller (VID 1532, PID 0F19) is detected twice by the kernel module.

z3ntu, 2020-10-27:
Hey, any update on this? I'd like to create a new release relatively soon (couple of weeks) and either this device should be fully supported or it shouldn't be included at all. Would it be possible that this gets fixed until then?

999eagle, 2020-10-27:
As I have no idea how to properly fix the same device showing up twice, maybe it would be an easy fix to check in the daemon and then just don't add the second device? It's a rather dirty fix imho, that's why I'd like to ask before implementing this.

PR #1399 (2021-01-17): Check for double interface in razeraccessory driver
This PR properly fixes the double sysfs interface provided for the Razer Kraken Kitty Edition.

Fixes #1286. See also #1312 for the temporary fix.
closed 2021-01-17 by z3ntu