openrazer/openrazer #1287: Razer Lancehead Tournament Edition Gunmetal - only scroll wheel and logo work for rgb control - Daemon complains about non-existent file

redgloboli, 2020-10-12:
### Distribution

Arch Linux Kernel 5.8.14

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Lancehead Tournament Edition does only show mousewheel and logo to set colors, left and right side is stuck in spectrum cycle mode.

I tried with razergenie and polychromatic

lsusb: ID 1532:0060 Razer USA, Ltd RZ01-0213 Gaming Mouse [Lancehead Tournament Edition]
firmware: 1.3
openrazer dkms: 2.6.0.r6.g2433ae8

dmesg:

[12310.682590] usb 1-13: Product: Razer Lancehead Tournament Edition
[12310.682592] usb 1-13: Manufacturer: Razer
[12310.684790] input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:1532:0060.000D/input/input43
[12310.684966] razermouse 0003:1532:0060.000D: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-13/input0
[12310.685781] input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.1/0003:1532:0060.000E/input/input44
[12310.739671] razermouse 0003:1532:0060.000E: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-13/input1
[12310.740122] input: Razer Razer Lancehead Tournament Edition as /devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.2/0003:1532:0060.000F/input/input45
[12310.795851] razermouse 0003:1532:0060.000F: input,hidraw6: USB HID v1.11 Keyboard [Razer Razer Lancehead Tournament Edition] on usb-0000:00:14.0-13/input2

razer_mount log:

Okt 12 22:40:53 mymachine razer_mount[103379]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103380]: Device_ID 1-13
Okt 12 22:40:53 mymachine razer_mount[103387]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103388]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103389]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103390]: Device_ID 1-13:1.0
Okt 12 22:40:53 mymachine razer_mount[103391]: Device_ID 1-13:1.1
Okt 12 22:40:53 mymachine razer_mount[103392]: Device_ID 1-13:1.2
Okt 12 22:40:53 mymachine razer_mount[103397]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103398]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103399]: Driver razermouse
Okt 12 22:40:53 mymachine razer_mount[103400]: Device_ID 0003:1532:0060.000D
Okt 12 22:40:53 mymachine razer_mount[103401]: Device_ID 0003:1532:0060.000E
Okt 12 22:40:53 mymachine razer_mount[103402]: Device_ID 0003:1532:0060.000F
Okt 12 22:40:53 mymachine razer_mount[103403]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000D/
Okt 12 22:40:53 mymachine razer_mount[103404]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000E/
Okt 12 22:40:53 mymachine razer_mount[103405]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000F/
Okt 12 22:40:53 mymachine razer_mount[103409]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000D/
Okt 12 22:40:53 mymachine razer_mount[103410]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000E/
Okt 12 22:40:53 mymachine razer_mount[103411]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.000F/

Daemon log:

2020-10-12 22:40:14 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 103236
2020-10-12 22:40:14 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-10-12 22:40:14 | razer                          | INFO     | Found device.0: 0003:1532:0060.0002
2020-10-12 22:40:14 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:1532:0060.0002/device_type, file is not owned by plugdev
2020-10-12 22:40:14 | razer                          | INFO     | Found device.0: 0003:1532:0C02.0005
2020-10-12 22:40:14 | razer.device0                  | INFO     | Initialising device.0 RazerGoliathusExtended
2020-10-12 22:40:14 | razer                          | INFO     | Serving DBus
2020-10-12 22:40:55 | razer                          | INFO     | Found valid device.1: 0003:1532:0060.000D
2020-10-12 22:40:55 | razer.device1                  | INFO     | Initialising device.1 RazerLanceheadTE
2020-10-12 22:43:31 | razer                          | INFO     | Stopping daemon on signal 15

this file /sys/devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:1532:0060.0002/device_type doesnt exist at all

i got /sys/devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/0003:1532:0060.000D/device_type and its owned by plugdev

If you need anything else let me know.

redgloboli, 2020-10-13:
ah ok, but the driver should be able to handle the side leds if the controlling software send the right commands?

no, the error doesn't appear anymore

2020-10-13 22:19:11 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 3424
2020-10-13 22:19:11 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-10-13 22:19:11 | razer                          | INFO     | Found device.0: 0003:1532:0060.0002
2020-10-13 22:19:11 | razer.device0                  | INFO     | Initialising device.0 RazerLanceheadTE
2020-10-13 22:19:11 | razer                          | INFO     | Found device.1: 0003:1532:0C02.0005
2020-10-13 22:19:11 | razer.device1                  | INFO     | Initialising device.1 RazerGoliathusExtended
2020-10-13 22:19:11 | razer                          | INFO     | Serving DBus

z3ntu, 2020-10-16:
> ah ok, but the driver should be able to handle the side leds if the controlling software send the right commands?

Exactly.

I'll close this issue then as the questions seem to be answered.

closed 2020-10-16 by z3ntu