openrazer/openrazer #1290: Basilisk settings persistent after uninstall

marcus-cr, 2020-10-13:
I had installed OpenRazer and set the colors to red, but these settings persist after uninstalling OpenRazer. The color stays red at all times (during POST, in Windows prior to Synapse running, in other Linux flavors not running OpenRazer. Attempted to reset the profile manually on the mouse (if this is even possible with this model) and within Synapse in Windows. I can change the color within Synapse but the red color previously set in OpenRazer is always consistent.

Basilisk

- Product number RZ01-02330100

- Model number RZ01-0233

z3ntu, 2020-10-14:
That sounds corrrect, we tell the devices to remember the effects we set so you can for example set them once and they get remembered by the device. If you want to change the color or effect just change it to a different effect in openrazer?

tildearrow, 2020-10-14:
This is because Synapse 3 no longer uses VARSTORE for storing the effect in the device, but rather does everything in software.
OpenRazer on the other hand still employs VARSTORE, which causes this behavioral difference.

Honestly, how long until #1149 gets merged? After this gets merged we can pretty much get rid of VARSTORE as we take the path towards doing it in software...

marcus-cr, 2020-10-16:
Well the issue I'm encountering is that I can't purge VARSTORE on my Basilisk mouse, even with Synapse. I managed to clear and revert my BlackWidow Elite to factory settings with Synapse by resetting the onboard profiles, but I can't do the same with my mouse.

It seems ridiculous that I have to use Synapse just to clear VARSTORE and revert to factory settings.

I'm just trying to undo all the settings/changes made with OpenRazer on my Basilisk mouse. I honestly don't know how to be more clear about that.

As a side note: it does seem that taking the path to make all changes in software would be best. Would help prevent issues like this even though my Basilisk mouse is considered supported, IIRC

lah7, 2020-10-16:
Is VARSTORE the same _thing_ that saves onboard profiles, brightness and last hardware effect in the firmware? If so, could packet captures from Synapse (2 or 3?) be used to safely support clear VARSTORE in the OpenRazer driver?

I ask as some later generation Razer hardware (e.g. Headphone Stand, HDK) seem to have hardware effects built-in to the firmware, even though Synapse 3 doesn't use them (being software only). The same could be the case for the Basilisk, so _unofficially_ clearing VARSTORE may be the only option since Synapse wouldn't have such option in their software?

marcus-cr, 2020-10-21:
Any resolution to this? Seems to still be delaying my boot up times. Keyboard non-functional at initial boot to enter BIOS.

z3ntu, 2020-10-22:
@marcus-cr We don't (and can't) do anything with the devices when Linux isn't booted so I'm not sure what you're expecting us to do.

marcus-cr, 2020-10-22:
@Zentetsu I booted into Ubuntu 20.04 and reinstalled OpenRazer. There isn't a way to reset the mouse to factory default.

z3ntu, 2020-10-22:
I'm not aware of any command we can send the device to do that, so I can't help you here.

Is there anything else you need help with?

closed 2021-06-20 by github-actions[bot]