openrazer/openrazer #1291: Adding Pop! Os as a download distro

Daggerpov, 2020-10-14:


z3ntu, 2020-10-14:
Hi, as far as I know Pop!_OS is based on Ubuntu so you should be able to use the normal Ubuntu instructions there. Does that work for you?

Daggerpov, 2020-10-14:
well, I did install it but I can't actually open the application. I search for it and there's an icon that shows up for both it and Polychromatic but they won't create a new window and open for whatever reason.

z3ntu, 2020-10-14:
Can you please run `polychromatic-controller` in a terminal and check the output?

Daggerpov, 2020-10-14:
[image]

z3ntu, 2020-10-14:
As you workaround you can do `sudo apt install python3-distro` , then it should launch fine. Does that make it work?

I've filed https://github.com/polychromatic/polychromatic/issues/289 so this can get fixed for everyone.

Daggerpov, 2020-10-14:
[image]

z3ntu, 2020-10-14:
`head -n1 /usr/bin/polychromatic-controller` and `which python3` please

Daggerpov, 2020-10-14:
[image]

lah7, 2020-10-14:
Try changing the shebang to `#!/usr/bin/python3` (you'll need to edit the file `/usr/bin/polychromatic-controller` and `/usr/bin/polychromatic-tray-applet` as root). It seems there's a local installation of Python and so it's failing to load the system-installed modules.

This won't be necessary in future versions of Polychromatic, since it's been changed to `#!/usr/bin/python3`.

Similar issue to https://github.com/polychromatic/polychromatic/issues/265#issuecomment-659548947

Daggerpov, 2020-10-14:
[image]

lah7, 2020-10-14:
Modifying files in `/usr/bin` requires superuser (root) privileges, you can do that with either:

    sudo nano /usr/bin/polychromatic-controller

And make the changes in the Terminal, then press <kbd>CTRL</kbd>+<kbd>X</kbd> then <kbd>Y</kbd> to save.

Or graphically - I believe since Pop_OS! is GNOME, the text editor would be **gedit**:

    sudo gedit /usr/bin/polychromatic-controller

Daggerpov, 2020-10-14:
Alright, the application works now and launches, but I'm unable to change the lighting effects for my Razer Viper Ultimate mouse or see the charging dock device for it. As for the first issue, this is the error raised in the terminal:
[image]
also, the python directories are different...

Daggerpov, 2020-10-14:
Wait nevermind, I needed to restart the daemon to resolve that first issue. So, I'm still wondering why the two python paths are different and why the charging dock won't show up as a device.

lah7, 2020-10-14:
The Python paths in that screenshot look correct for an Ubuntu-based distro (`/usr/lib/python3/dist-packages/`). Most other distros (and custom installs) normally use `site-packages`.

The charging dock doesn't appear as a separate device, but OpenRazer does support changing its effect/colour, there are some scripts in this repo that may work:

    scripts/razer_mouse/driver/set_charging_colour.py
    scripts/razer_mouse/driver/set_charging_effect.py

Unfortunately, Polychromatic doesn't support charging docks _yet_.

Daggerpov, 2020-10-14:
how do I navigate to these scripts in the repo?

lah7, 2020-10-14:
Scratch that part about the charging docks, I'm unfamiliar with them. It should appear as a separate device (like `Razer Lancehead Wireless (Receiver)`) and therefore should work in Polychromatic (no need for those scripts)

If you haven't already since installing OpenRazer, you should restart the computer so the device is properly bound to the driver.

Daggerpov, 2020-10-14:
Well, even when I restart my computer it doesn't show up in the overview on devices. However, it does light up.

nightsky30, 2020-10-14:
@Daggerpov Fairly certain the Viper Ultimate and Basilisk Ultimate mice come with the same dock (or at least it seems issues tracking the device support for those mice referenced the same dock PID). If it's the Razer charging dock with PID 007e, then it looks like support is being added in PR #1152 along with the Basilisk Ultimate.

Daggerpov, 2020-10-16:
Yes, they both come with the same charging dock. So you're basically just telling me to wait until it's fixed in a later version of this software?

nightsky30, 2020-10-16:
> Yes, they both come with the same charging dock. So you're basically just telling me to wait until it's fixed in a later version of this software?

I'm simply saying device support appears to be in the works (to answer as to why it isn't working for you now). I don't know if/when that PR will be merged. Does not seem the author is finishing it any time soon.

Waiting is one option, yes. Building that [branch](https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint) for your own personal use is another. And I'm sure testing and feedback for device support would be appreciated. It seems getting the brightness value for the charger isn't working (makes me wonder if that is another wait time issue).

Daggerpov, 2020-10-17:
alright, I was just clarifying. I wasn't trying to bash the author or talk badly about the software so don't take it as that.

closed 2020-10-17 by Daggerpov