openrazer/openrazer #1293: Custom Effects Colors

mind-overflow, 2020-10-16:
I was unable to find any other open or closed issue about this, so I'm opening one. I apologize in case this has already been answered.
As you probably know, the effects that support custom colors are, currently, Static, Starlight, Ripple, Reactive and Breath.
Spectrum and Wave currently do not support setting custom colors.
Why is that? What is missing to make this possible, and are there any plans for it?

tildearrow, 2020-10-16:
This is due to the fact that OpenRazer only uses hardware effects (which do not have color/speed parameters for Spectrum and Wave), whereas Synapse 3 does the effects in software, which allows for this.

mind-overflow, 2020-10-16:
Understood. So this will never be possible, as I assume it's highly inefficient, or are there any plans for it?
I can help reverse engineering the USB commands if needed.

tildearrow, 2020-10-16:
The base for it already exists (being able to set custom frames), and in fact this is what the Ripple effect uses.
However, what is missing is the rest of the software effects.

From what I have heard, z3ntu is working on a rewrite of OpenRazer called razer_test, which has some basic software effects; however, it is far from usable.

lah7, 2020-10-16:
Using custom frames can be done with the Python library API:

```python
import openrazer.client
d = openrazer.client.DeviceManger()
a = d.devices[0]
print(a.name)               # Razer BlackWidow Chroma
print(a.fx.advanced.cols)   # 22
print(a.fx.advanced.rows)   # 6
a.fx.advanced.matrix[0,0] = [255,255,255]   # ESC key is white
a.fx.advanced.draw()
```

Examples could be contributed to the [`scripts/daemon` folder](https://github.com/openrazer/openrazer/tree/master/scripts/daemon).

---

[Polychromatic](https://polychromatic.app/), one of the frontends that utilises OpenRazer will be overhauling the custom effects functionality pretty soon. :crossed_fingers: Currently it only uses custom frames to light up **static** keys for games/applications. It will take some time to complete for more complex animations and layers, but hopefully before the end of the year. :turtle:

mind-overflow, 2021-05-23:
the issue is not stale, it's still a work in progress.

j-silver, 2021-07-25:
Any progress on this? Using the Python API is cool, but being able to create custom effects with polychromatic would be even more... :-)

lah7, 2021-07-25:
Nothing is being worked on at an OpenRazer level that I know of. The API is there and any one is welcome to contribute examples. OpenRazer is pretty much focused on exposing hardware features (Ripple is an exception)

Over on Polychromatic's side, the latest codebase has the ability to create frame-by-frame animations or something static, so the power is there to create a custom wave using the colouring tools. The plan for a more fancier (complex) layered and software effects will be worked on much later in the year.

mind-overflow, 2021-07-31:
@lah7 This is great news! Thanks for sharing :)  
Just a quick question:  
  
>  The plan for a more fancier (complex) layered and software effects will be worked on much later in the year.
  
You are still talking about Polychromatic, right? Like implementing a GUI similar to the original one where you can create effects dynamically?

lah7, 2021-08-01:
Yes, I'm talking about Polychromatic. I'm thinking of a layer-based approach since the brief time I used/seen Razer Synapse, it seemed pretty confusing and complicated to me. I would've had something built sooner, but I seem to work in _valve time_ 🙂

mind-overflow, 2021-08-03:
> Yes, I'm talking about Polychromatic. I'm thinking of a layer-based approach since the brief time I used/seen Razer Synapse, it seemed pretty confusing and complicated to me.

Awesome! I don't know what version of Synapse have you tried, but Synapse 3 (although admittedly not very user friendly) *is* layered. You chose an effect and add it on top of another one. After a few minutes of doing stuff randomly, you finally understand how it works tho.  
  
>  I would've had something built sooner, but I seem to work in _valve time_
  
Haha, I appreciate the humor. Don't worry, what we have right now is already great given that it all comes from reverse engineering and is 100% open source. Take your time and when stuff will be ready, we'll be even happier :) I'd love to contribute developing but unfortunately I'm in a moment where I don't really have much time.  
  
Thank you for everything!