openrazer/openrazer #1296: Problem communicating with the daemon for DeathAdder 3.5G

AurelC2G, 2020-10-17:
### Distribution

Ubuntu 20.04 LTS

```
$ polychromatic-controller --version
Polychromatic v0.3.12-stable
```
```
$ openrazer-daemon --version
openrazer-daemon 2.8.0
```

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,

I have two razer devices: a BlackWidow Stealth keyboard, and a DeathAdder 3.5G mouse.
Both show up in the polychromatic controller UI, as well as when running `polychromatic-controller --print-device-info`.
I can control the keyboard with no issues via the UI.

However, when I select the mouse in the UI, the following error message appears:

[image]

When running `polychromatic-controller -v` on the command line, I get some additional info:
```
There was a problem communicating with the daemon for this device.
Exception: not enough values to unpack (expected 2, got 1)
```

This error shows up again whenever I change the dpi setting for the mouse, and the change seems to have no effect.
On the other hand, changing the lighting options for the mouse does work as expect.

```
# Device Info
Name: Razer DeathAdder 3.5G
Type: mouse
 
# Device Capabilities
{'name': True, 'type': True, 'firmware_version': True, 'serial': True, 'brightness': False, 'macro_logic': False, 'keyboard_layout': False, 'lighting': False, 'lighting_breath_single': False, 'lighting_breath_dual': False, 'lighting_breath_triple': False, 'lighting_breath_random': False, 'lighting_wave': False, 'lighting_reactive': False, 'lighting_none': False, 'lighting_spectrum': False, 'lighting_static': False, 'lighting_starlight_single': False, 'lighting_starlight_dual': False, 'lighting_starlight_random': False, 'lighting_ripple': False, 'lighting_ripple_random': False, 'lighting_pulsate': False, 'lighting_led_matrix': False, 'lighting_led_single': False, 'lighting_logo': True, 'lighting_logo_active': True, 'lighting_logo_blinking': False, 'lighting_logo_brightness': False, 'lighting_logo_pulsate': False, 'lighting_logo_spectrum': False, 'lighting_logo_static': False, 'lighting_logo_none': False, 'lighting_logo_reactive': False, 'lighting_logo_wave': False, 'lighting_logo_breath_single': False, 'lighting_logo_breath_dual': False, 'lighting_logo_breath_random': False, 'lighting_scroll': True, 'lighting_scroll_active': True, 'lighting_scroll_blinking': False, 'lighting_scroll_brightness': False, 'lighting_scroll_pulsate': False, 'lighting_scroll_spectrum': False, 'lighting_scroll_static': False, 'lighting_scroll_none': False, 'lighting_scroll_reactive': False, 'lighting_scroll_wave': False, 'lighting_scroll_breath_single': False, 'lighting_scroll_breath_dual': False, 'lighting_scroll_breath_random': False, 'lighting_left': False, 'lighting_left_active': False, 'lighting_left_brightness': False, 'lighting_left_spectrum': False, 'lighting_left_static': False, 'lighting_left_none': False, 'lighting_left_reactive': False, 'lighting_left_wave': False, 'lighting_left_breath_single': False, 'lighting_left_breath_dual': False, 'lighting_left_breath_random': False, 'lighting_right': False, 'lighting_right_active': False, 'lighting_right_brightness': False, 'lighting_right_spectrum': False, 'lighting_right_static': False, 'lighting_right_none': False, 'lighting_right_reactive': False, 'lighting_right_wave': False, 'lighting_right_breath_single': False, 'lighting_right_breath_dual': False, 'lighting_right_breath_random': False, 'lighting_backlight': False, 'lighting_backlight_active': False, 'lighting_profile_led_red': False, 'lighting_profile_led_green': False, 'lighting_profile_led_blue': False, 'poll_rate': True, 'dpi': True, 'available_dpi': True, 'battery': False}
```

z3ntu, 2020-10-17:
Sounds like this is already tracked in https://github.com/polychromatic/polychromatic/issues/209

closed 2020-10-17 by z3ntu

lah7, 2020-10-17:
Polychromatic v0.3.12 doesn't properly support this mouse due to it having 1 DPI axis whereas the application expects two (X & Y).

The [development builds of this application](https://github.com/polychromatic/polychromatic/issues/263) have addressed the problem.