openrazer/openrazer #1298: Support for Charging Dock

Daggerpov, 2020-10-17:
### Device Info

* Name: I use this with my Razer Viper Ultimate mouse and apparently it's also used with the Basilisk Ultimate and on the back it's called "Charging Dock"
* Product Number: RC30-03050200-0000
* Model Number: rc30-030502

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 003 Device 004: ID 1532:007e Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x007e 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 171 lines cut ...]
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/

![image](https://user-images.githubusercontent.com/53918934/96338304-ff39a100-105a-11eb-9a63-d0efe68c76d6.png)

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

Daggerpov, 2020-10-17:
Sorry if I left out some info, but I'd rather leave those blank then spew about something I know nothing of, like what it'd look like with Synapse or these "Packet Captures".

z3ntu, 2020-10-18:
This will be implemented in #1152

Daggerpov, 2020-10-18:
Sounds good!

lah7, 2021-10-31:
This was implemented in #1321

closed 2021-10-31 by lah7