openrazer/openrazer #1302: Razer Kraken Tournament Edition - Gaming Headset dont detected in linux

jpVm5jYYRE1VIKL, 2020-10-22:
### Distribution

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal

### Support

- [ ] Have you tried rebooting?
Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
Yes

Razer Kraken Tournament Edition - Gaming Headset don`t detected in linux during boot. After fresh boot Razer Kraken not in list of usb devices. And start to work only after removal usb and placing it back. 

Ubuntu system booted and usb  Razer Kraken stay connected during boot: 

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c531 Logitech, Inc. C-U0007 [Unifying Receiver]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 010: ID 046d:c229 Logitech, Inc. G19 Gaming Keyboard Macro Interface
Bus 001 Device 009: ID 046d:c228 Logitech, Inc. G19 Gaming Keyboard
Bus 001 Device 008: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 001 Device 007: ID 1b1c:1c0d Corsair                       
Bus 001 Device 006: ID 046d:c332 Logitech, Inc. G502 Proteus Spectrum Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Ubuntu system booted and usb  Razer Kraken physically disconnected and connected again after boot: 

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c531 Logitech, Inc. C-U0007 [Unifying Receiver]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 1532:0520 Razer USA, Ltd Razer USB Audio Controller
Bus 001 Device 010: ID 046d:c229 Logitech, Inc. G19 Gaming Keyboard Macro Interface
Bus 001 Device 009: ID 046d:c228 Logitech, Inc. G19 Gaming Keyboard
Bus 001 Device 008: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 001 Device 007: ID 1b1c:1c0d Corsair                       
Bus 001 Device 006: ID 046d:c332 Logitech, Inc. G502 Proteus Spectrum Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

z3ntu, 2020-10-22:
Can you try uninstalling OpenRazer and see if the problem persists?

jpVm5jYYRE1VIKL, 2020-10-22:
Yes and did it even multiply times.

z3ntu, 2020-10-22:
Generally the usb device must be detected by the Linux kernel and openrazer only can communicate with the device after it has been found by the kernel (and openrazer also isn't involved in finding the device), so this must be an issue with your setup or a bug in Linux.

closed 2020-10-22 by z3ntu

jpVm5jYYRE1VIKL, 2020-10-22:
Are you serious . It is a problem only with razor . I can believe that it is a bug in razor than in linux usb handling. Another devices in linux working fine and only with razor problem. But ok.Thanks for "excellent" bug handling.

gierappa, 2021-12-16:
I have the same problem. Has anyone managed to solve it?