openrazer/openrazer #1304: Request to add a new client to the website

brett-smith, 2020-10-23:
I am the author of a new OpenRazer client application, [Snake](http://bithatch.co.uk/snake.html), and wondered if it could be added to the website in your list of clients.

It can be considered beta quality, and I am ready for others to start testing it out now.  The best way to install currently is using the self extracting .sh installer. 

It makes use of JavaFX, and is probably more like Polychromatic in style than it is either razerCommand or razerGenie, but with it's own twist.

 * It is self updating, and no dependencies other than OpenRazer
 * Themes, animations, effects and window transparency for looks
 * Supports all effects and zones
 * Supports multiple devices
 * Has a tray icon
 * I've tried to support as many devices as I can (thanks for the fake drivers!), but only actually own a Razer mouse so please report any hardware specific issues, particularly if you know it works in other clients.

The one area not quite up to standard is macro support, but I am awaiting the changes in this area that seem to be under development before committing much more time to the current implementation.

GitHub project here - https://github.com/bithatch/snake

lah7, 2020-10-23:
Thanks for sharing - nice to see a Java implementation! :coffee: 

Sure, we'll add the client to the list in the README as well as the website. Would you prefer us to link to the website or the GitHub project?

Also, a word to the wise, I'd strongly recommend changing the logo. :wink:

brett-smith, 2020-10-23:
I'd prefer a link to the website please.

And yeh, noted on the logo. It was kind of a place holder. A original application, icon is on the list of things to do soon. I wanted to make it part of the theme. This is now done, so I just need to actually draw something. Expect it to change over the  weekend.

commit 486d5b83 (2020-10-23): Add Snake java application to README

Closes #1304

https://github.com/bithatch/snake
closed 2020-10-23 by lah7