openrazer/openrazer #1306: Razer Blade Advanced 2020 - Keyboard Not Working

dcrdev, 2020-10-23:
### Distribution

Fedora 31

### Support

- [x ] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Just got my Razer Blade 15 Advanced Model today (2020)  and am attempting to get open razer working. Dkms modules have successfully built:
```
[root@hell01-mw01 dominic]# dkms status
openrazer-driver, 2.8.0, 5.8.15-201.fc32.x86_64, x86_64: installed
```

Looks as though, the keyboard isn't being picked up:
```
[root@hell01-mw01 dominic]# journalctl -b | grep -i razer_mount
Oct 23 16:22:52 hell01-mw01 razer_mount[1390]: Driver razermouse
Oct 23 16:22:52 hell01-mw01 razer_mount[1389]: Driver razermouse
Oct 23 16:22:52 hell01-mw01 razer_mount[1391]: Device_ID 0003:1532:0078.0005
Oct 23 16:22:52 hell01-mw01 razer_mount[1392]: Device_ID 0003:1532:0078.0006
Oct 23 16:22:52 hell01-mw01 razer_mount[1393]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0005/
Oct 23 16:22:52 hell01-mw01 razer_mount[1394]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0006/
Oct 23 16:22:52 hell01-mw01 razer_mount[1401]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0005/
Oct 23 16:22:52 hell01-mw01 razer_mount[1402]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0006/
Oct 23 16:22:52 hell01-mw01 razer_mount[1410]: Driver razermouse
Oct 23 16:22:52 hell01-mw01 razer_mount[1411]: Device_ID 0003:1532:0078.0004
Oct 23 16:22:52 hell01-mw01 razer_mount[1412]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0004/
Oct 23 16:22:52 hell01-mw01 razer_mount[1414]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0078.0004/
```

Which is odd:
```
[dominic@hell01-mw01 ~]$ lsmod | grep razer
razerkbd               77824  0
razermouse             94208  0
```

My hardware should - be supported right (it is according to the readme) :
```
[dominic@hell01-mw01 ~]$ lsusb | grep 1532
Bus 001 Device 006: ID 1532:0253 Razer USA, Ltd RZ09-0330, Gaming Laptop [Blade 15 Advanced (Early 2020)]
Bus 001 Device 004: ID 1532:0078 Razer USA, Ltd Viper (wired)
Bus 001 Device 015: ID 1532:0520 Razer USA, Ltd Kraken Tournament Edition
```

So looks like the device isn't mounting for whatever reason - possibly udev related? Not quite sure where to begin debugging this, so thought I'd ask.

Also worth mentioning that I have a razer external mouse, which IS working.

Cheers

lah7, 2020-10-23:
It looks like your device (`1532:0253`) isn't on the [stable branch](https://github.com/openrazer/openrazer/tree/stable) yet, so it _doesn't exist_ until the next version (2.9.0).

In the meantime, try building a package based on the `master` branch. I'm terribly unfamiliar with Fedora (sorry!) on how to do this.

dcrdev, 2020-10-24:
@lah7 thanks - I missed that, this indeed does work with the current master.

If anyone lands on this issue in future, you can rebuild against master fairly easily in fedora:
```bash
$ mkdir openrazer-meta-git486d5b8 && cd openrazer-meta-git486d5b8
$ wget https://ftp.gwdg.de/pub/opensuse/repositories/hardware:/razer/Fedora_32/src/openrazer-meta-2.8.0-2.1.src.rpm
$ rpm2cpio ./*.src.rpm | cpio -i
$ sed -i '/#define gitcommit/c\%define gitcommit 486d5b83ffde2b192f06da647e696347339b13cf' *.spec
Vrm ./{*.tar.xz,*.src.rpm}
$ spectool -g *.spec
$ sudo dnf install mock
$ export ME=`whoami`
$ sudo -E usermod -aG mock "$ME" 
$ mock --buildsrpm --spec *.spec  --sources *.tar.gz --resultdir .
$ mock --rebuild *.src.rpm  --sources *.tar.gz --resultdir .
$ sudo dnf install *.noarch.rpm
```

closed 2020-10-24 by dcrdev