openrazer/openrazer #1307: Support for basilisk x hyperspeed

BinaryJava, 2020-10-24:
### Device Info

* Name: Basilisk X Hyperspeed
* Product Number: RC30-03150100-0000
* Model Number: RC30-031501

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
Bus 001 Device 011: ID 1532:0083 Razer USA, Ltd Razer Goliathus Extended Chroma
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0083 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 306 lines cut ...]
lrwxrwxrwx 1 root root   6 Oct 24 21:47 usb-©Microsoft_Corporation_Controller_07491BE-joystick -> ../js0
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-MSI_MYSTIC_LIGHT_A02018080702-event-if00 -> ../event24
lrwxrwxrwx 1 root root  11 Oct 24 21:47 usb-Razer_Razer_Basilisk_X_HyperSpeed-event-if01 -> ../event258
lrwxrwxrwx 1 root root  11 Oct 24 21:47 usb-Razer_Razer_Basilisk_X_HyperSpeed-event-mouse -> ../event256
lrwxrwxrwx 1 root root  11 Oct 24 21:47 usb-Razer_Razer_Basilisk_X_HyperSpeed-if01-event-kbd -> ../event257
lrwxrwxrwx 1 root root  11 Oct 24 21:47 usb-Razer_Razer_Basilisk_X_HyperSpeed-if02-event-kbd -> ../event261
lrwxrwxrwx 1 root root   9 Oct 24 21:47 usb-Razer_Razer_Basilisk_X_HyperSpeed-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Goliathus_Extended_Chroma-event-if01 -> ../event20
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Goliathus_Extended_Chroma-event-mouse -> ../event18
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Goliathus_Extended_Chroma-if01-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Goliathus_Extended_Chroma-if02-event-kbd -> ../event21
lrwxrwxrwx 1 root root   9 Oct 24 21:47 usb-Razer_Razer_Goliathus_Extended_Chroma-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Huntsman-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Huntsman-if01-event-kbd -> ../event23
lrwxrwxrwx 1 root root  10 Oct 24 21:47 usb-Razer_Razer_Huntsman-if02-event-mouse -> ../event26
lrwxrwxrwx 1 root root   9 Oct 24 21:47 usb-Razer_Razer_Huntsman-if02-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-10-26:
The device is already supported in the master branch (#1129) and will be included in the next release. You can install from the daily ppa, -git package from AUR or build from source in the meantime.

closed 2020-10-26 by z3ntu

szul, 2020-11-18:
@z3ntu Is this in the 2.9 daemon release?

z3ntu, 2020-11-18:
Yes

szul, 2020-11-18:
Hmm... Does it only work with the USB dongle? I'm using BlueTooth and it doesn't seem to be recognizing the mouse. Recognizes my headphones just fine though.

z3ntu, 2020-11-18:
@szul Only the usb id `1532:0083` is supported. @bno1 can you chime in here?

bno1, 2020-11-18:
@z3ntu The driver doesn't work with the mouse in bluetooth mode and I'm not sure what needs to be done.

I found the device ID (`0082`) in `bluetoothctl`:
```
$ bluetoothctl info F2:12:E5:82:5A:B9
Device F2:12:E5:82:5A:B9 (random)
        Name: Basilisk X HyperSpeed
        Alias: Basilisk X HyperSpeed
        Appearance: 0x03c2
        Icon: input-mouse
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (52401523-f97c-7f90-0e7f-6c6f4e36db1c)
        Modalias: usb:v1532p0082d0001
        Battery Percentage: 0x17 (23)
```

I tried adding `0082` to my build but the driver doesn't seem to catch it:
```
[  757.966862] input: Basilisk X Mouse as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input42
[  757.967036] input: Basilisk X Consumer Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input43
[  757.967120] input: Basilisk X System Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input44
[  757.967197] input: Basilisk X as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input45
[  757.967271] input: Basilisk X Keyboard as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input46
[  757.967386] hid-generic 0005:1532:0082.0008: input,hidraw7: BLUETOOTH HID v0.01 Mouse [Basilisk X] on a8:7e:ea:d7:32:5e
```

bno1, 2020-11-18:
Maybe something needs to be changed in `99-razer.rules` to make it use the `razermouse` driver for the bluetooth devices too.

szul, 2020-11-18:
@bno1 Should I create a new issue for this then?

bno1, 2020-11-18:
@szul Yes. Also add a link to this comment, might be helpful: https://github.com/openrazer/openrazer/pull/1203#issuecomment-663992279.

c2xusnpq6, 2020-12-13:
https://github.com/openrazer/openrazer/issues/1047#issuecomment-743935791
https://github.com/openrazer/openrazer/issues/1047#issuecomment-743936909

z3ntu, 2020-12-13:
@c2xusnpq6 Please do not spam on closed issues, thanks.