openrazer/openrazer #1310: Support for the Razer Huntsman Mini

Tochey, 2020-10-26:
Can you add support for the Razer Huntsman Mini. I love what you guys are doing

z3ntu, 2020-10-26:
Can you please check whether your device has the usb id `1532:0257`? `lsusb | grep 1532` should tell you

Tochey, 2020-10-26:
Where can I find the usb id? I looked on the back and nothing there

z3ntu, 2020-10-26:
You can run `lsusb | grep 1532` in a terminal.

Tochey, 2020-10-26:
tried running it on a Mac terminal but the command was not found 
[image]
Does this image help?

z3ntu, 2020-10-26:
Yes, this is a duplicate of #1230 then, but I'm not sure why you're asking for support in a Linux driver when you're using macOS?

closed 2020-10-26 by z3ntu