openrazer/openrazer #1313: What has happened to getRazerURLs?

brett-smith, 2020-10-29:
### Distribution

N/A

### Support

- [ N/A] Have you tried rebooting?
- [ N/A] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I notice that in the current master branch that `getRazerURLs` is gone. 

Trying to track down why this might be, I came across this commit. https://github.com/openrazer/openrazer/commit/83d275d8584fcc71b8eab299c799240faa5d05a3

The text for this says 'Frontends have only been using one image, so only one device image needs
to be defined.', but this is not entirely true, at least now :)

Snake is using the **top_img** down image for most 'icon' type usage, but I was also using **perspective_img** which I  faded and desaturated for the background image in the UI for a specific device.

I also had future plans for the side image, although nothing concrete yet.

So if the only reason is clients don't use it, could I request that this be re-instated. If you have other reasons, such as a maintenance burden, I am happy to investigate a client-specific solution to get these images back.

z3ntu, 2020-10-29:
Hi, the old urls have been deprecated for over a year (since https://github.com/openrazer/openrazer/commit/aeabb468fe1ec700ac4cc1647cd09b93a26daeed).

The reason mostly is that it's quite hard to find fitting images for all devices as Razer isn't very consistent in the images they provide for their devices. We sometimes even have problems finding a single picture that fits well.

But I don't think `getRazerURLs` is quite gone yet, it should be a wrapper around the getDeviceImage api, returning the same picture for all types of course. 

https://github.com/openrazer/openrazer/blob/de0f266766cd7e6efdf3235e8b2649657612cb02/daemon/openrazer_daemon/hardware/device_base.py#L280-L286

brett-smith, 2020-10-29:
Thanks for the explanation. I had a feeling it was probably a bit of a pain to get those images in the first place.

I had developed Snake against the stable version that has the full compliment of images. On first release, this was flagged up to me in the first issue reported against my client. They must have had a newer build from somewhere, or was running from checked out source from before the above shim was added, so Snake crashed. 

Anyway, i'll reconsider how important this is to me if its that inconsistent. I had hoped to use the images as a lazy way to add some visual interest. 

For some features I have planned, I am probably going to need some kind of client-side device metadata anyway, so adding any additional images here is probably what will happen.

closed 2021-06-20 by github-actions[bot]