openrazer/openrazer #1315: Support for [Macs]

Razerviper, 2020-10-29:
### Device Info

* Name: <Macs>
Are you adding support for macs soon? I really need synapse 3 for mac or something like it. 
If not, thank you anyways.

z3ntu, 2020-10-29:
Please read any of the issues found when searching for `macOS` in the issue search... https://github.com/openrazer/openrazer/issues?q=macos

closed 2020-10-29 by z3ntu

lah7, 2020-10-29:
> Simply put, it's not possible for OpenRazer in its current form to run on Windows/Mac because it depends on a Linux-only driver. There's @z3ntu's razer_test which handles everything in user space, eliminating the need for a driver. It's a work-in-progress, and is already making progress to run on Windows/Mac.
>
> Eventually that project will supersede OpenRazer 2.x, so hopefully some day.

From here: https://github.com/openrazer/openrazer/issues/1038#issuecomment-598193639.