openrazer/openrazer #1318: Support for BlackWidow V3 Tenkeyless

valotas, 2020-10-31:
### Device Info

* Name: `BlackWidow V3 Tenkeyless`
* Product Number: `RZ03-03490100-R3M1`
* Model Number: `RZ03-0349`

### Device Information
```
$ lsusb -d 1532: -v
```
```
Bus 003 Device 008: ID 1532:0a24 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a24 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```
```
$ ls -lah /dev/input/by-id/
```
```
total 0
drwxr-xr-x 2 root root 260 Oct 31 13:00 .
drwxr-xr-x 4 root root 860 Oct 31 13:00 ..
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-046d_HD_Pro_Webcam_C920_AB8FBA2F-event-if00 -> ../event25
lrwxrwxrwx 1 root root  10 Oct 31 12:59 usb-CKFGH10F442030026752_Integrated_Webcam_HD_200901010001-event-if00 -> ../event21
lrwxrwxrwx 1 root root   9 Oct 31 12:59 usb-ELAN_Touchscreen-event-if00 -> ../event8
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event24
lrwxrwxrwx 1 root root   9 Oct 31 13:00 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-if01 -> ../event27
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-if03 -> ../event31
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-event-kbd -> ../event22
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if01-event-kbd -> ../event26
lrwxrwxrwx 1 root root  10 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if02-event-mouse -> ../event30
lrwxrwxrwx 1 root root   9 Oct 31 13:00 usb-Razer_Razer_BlackWidow_V3_Tenkeyless-if02-mouse -> ../mouse4
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

valotas, 2020-11-01:
I am trying to build/install the project from source in order to create a PR following the instructions found [here](https://github.com/openrazer/openrazer/wiki/Building-a-package) but I get the following:

```
(Reading database ... 245566 files and directories currently installed.)
Preparing to unpack .../openrazer-daemon_2.8.0-0_all.deb ...
Unpacking openrazer-daemon (2.8.0-0) over (2.8.0-0) ...
Preparing to unpack .../openrazer-doc_2.8.0-0_all.deb ...
Unpacking openrazer-doc (2.8.0-0) over (2.8.0-0) ...
Preparing to unpack .../openrazer-driver-dkms_2.8.0-0_all.deb ...

-------- Uninstall Beginning --------
Module:  openrazer-driver
Version: 2.8.0
Kernel:  5.4.0-52-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

razerkbd.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.0-52-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razermouse.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.0-52-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razerkraken.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.0-52-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razeraccessory.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.0-52-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.8.0
completely from the DKMS tree.
------------------------------
Done.
Unpacking openrazer-driver-dkms (2.8.0-0) over (2.8.0-0) ...
Preparing to unpack .../openrazer-meta_2.8.0-0_all.deb ...
Unpacking openrazer-meta (2.8.0-0) over (2.8.0-0) ...
Preparing to unpack .../python3-openrazer_2.8.0-0_all.deb ...
Unpacking python3-openrazer (2.8.0-0) over (2.8.0-0) ...
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on python3-pyudev; however:
  Package python3-pyudev is not installed.
 openrazer-daemon depends on python3-setproctitle; however:
  Package python3-setproctitle is not installed.
 openrazer-daemon depends on python3-daemonize (>= 2.4.0); however:
  Package python3-daemonize is not installed.
 openrazer-daemon depends on xautomation; however:
  Package xautomation is not installed.

dpkg: error processing package openrazer-daemon (--install):
 dependency problems - leaving unconfigured
Setting up openrazer-doc (2.8.0-0) ...
Setting up openrazer-driver-dkms (2.8.0-0) ...
Loading new openrazer-driver-2.8.0 DKMS files...
Building for 5.4.0-52-generic
Building initial module for 5.4.0-52-generic
Done.

razerkbd.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-52-generic/updates/dkms/

razermouse.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-52-generic/updates/dkms/

razerkraken.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-52-generic/updates/dkms/

razeraccessory.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-52-generic/updates/dkms/

depmod...

DKMS: install completed.

ADD YOUR USER TO THE PLUGDEV GROUP: 'sudo gpasswd -a $USER plugdev'
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-daemon (= 2.8.0-0); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package openrazer-meta (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 2.8.0-0); however:
  Package openrazer-daemon is not configured yet.
 python3-openrazer depends on python3-numpy; however:
  Package python3-numpy is not installed.

dpkg: error processing package python3-openrazer (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.9.1-1) ...
Errors were encountered while processing:
 openrazer-daemon
 openrazer-meta
 python3-openrazer
```

am I missing something? Sorry If my question sounds silly but it is my first try into deb packaging and kernel driver

z3ntu, 2020-11-01:
I think you need to manually install `python3-numpy`, that's what I get from the error message

ArangoGutierrez, 2020-12-05:
Hey I just got a BlackWidow V3 Tenkeyless is there any progress on this issue? or anyone has a working solution?

valotas, 2020-12-06:
As sad as it sounds, I solved my problem by replacing the keyboard :(.

Having a look at the code though, it does not look like that hard to add support for another device id.

ArangoGutierrez, 2020-12-06:
Could you point me to were to start, I might try to add the device during holiday season.
Thanks

valotas, 2020-12-06:
@ArangoGutierrez you could start with [this branch of mine](https://github.com/valotas/openrazer/tree/feat/blackwidow-v3-tenkeyless) which has not been tested at all

ArangoGutierrez, 2020-12-06:
Thanks a lot @valotas , will git it a try and report back here

BitBangingBytes, 2021-01-02:
Also could use support, just got this keyboard for my Ubuntu machine.

NotNocky, 2021-01-28:
Happy to pcap stuff if i could get more info on what you need ? got the vm and wireshark set up now just dont want to flood with useless info.

NotNocky, 2021-01-28:
[image]
[image]
[image]
[image]
[image]

some synapse screenshots i got 

as well as the current wireshark set up i have, bit lost :sweat: 
[image]

z3ntu, 2021-01-31:
@NotNocky Just go through the different lighting options (don't need to spend much time there), change brightness a bit; these are probably the interesting parts.

NotNocky, 2021-02-03:
> @NotNocky Just go through the different lighting options (don't need to spend much time there), change brightness a bit; these are probably the interesting parts.

ah woops i forgot to check this email, so would that be in a single recording? from boot or what. is the following alright? start wm, open synapse, start recording, go thru all lighting options here https://user-images.githubusercontent.com/65059771/106093615-7c65cd80-6195-11eb-8b4c-f95e4b68dbc8.png , adjust brightness, stop recording

z3ntu, 2021-02-03:
@NotNocky yeah sounds good. You can also start wireshark before starting Windows, there are some unusual packets sent on Synapse startup for some devices so it might be nice to include them.

flopglop, 2021-02-18:
@z3ntu Here is my pcap. I have never used wireshark before so I hope I didn't mess anything up.
I have also attached the output of `lsusb -d 1532: -v` and `ls -lah /dev/input/by-id/` in case you need it.
If you need anything else or if I did something wrong don't hesitate to tell me.
[Blackwidow V3 tkl.zip](https://github.com/openrazer/openrazer/files/6003993/Blackwidow.V3.tkl.zip)

syndalis, 2021-02-23:
I also have the BlackWidow v3 TKL and attached is my pcap.

[bw3tkl.zip](https://github.com/openrazer/openrazer/files/6030896/bw3tkl.zip)

Let me know how I can help.

z3ntu, 2021-02-27:
Both pcaps have the info, thanks.

If someone wants to look into implementing this, use the `razer_chroma_extended_matrix_effect_*` functions with ZERO_LED (you can also try BACKLIGHT_LED if that works), `razer_chroma_extended_matrix_brightness` for brightness (also ZERO_LED). For custom frame use `razer_chroma_extended_matrix_set_custom_frame` and `MATRIX_DIMS = [6, 18]` . For reference look at recent commits for razerkbd.

madgeniusmusic, 2021-02-28:
Hi.

I've just got this keyboard and would appreciate it if I could get this working. Has there been any progress made?

diogoverardi, 2021-03-04:
hi Guys, was there any progress made on this? is there anything that I can help with?

pavlos256, 2021-05-14:
> This device is more or less a copy of Huntsman TE

See https://github.com/1kc/razer-macos/pull/285. According to that, it's the same as 1532:0243 which is already supported?

NotNocky, 2021-06-04:
Any updates here? / any work around to patch this in ourselves?

willeccles, 2021-06-04:
I'll give implementing this a shot. I'm not familiar with the python library, but it seems fairly straightforward as device drivers go.

willeccles, 2021-06-11:
If anyone with this keyboard could help out, I'd appreciate it if someone could test my patch. I have no idea whether or not it works, and some more data would be good. Here's my branch: https://github.com/willeccles/openrazer/tree/blackwidow-v3-tk

closed 2021-06-30 by z3ntu

sgofferj, 2021-07-01:
Works here, except the app/daemon doesn't recognize when game mode is activated on the keyboad with Fn+F10.

atloo, 2021-09-02:
Works here, except, thanks!

PR #1523 (2021-06-30): Support for BlackWidow V3 Tenkeyless
Given there has been no progress on #1318 in quite some time and one of my friends has this board, she asked me to take a crack at it. I've done the best I can without the hardware in front of me, so this is WIP until it is confirmed working on her machine.

Fixes #1318