openrazer/openrazer #1319: Support for BlackWidow V3 (Full Size)

dustincalhoun, 2020-11-02:
### Device Info

* Name: Razer BlackWidow V3
* Product Number: RZ03-0354
* Model Number: RZ03-03540200-R3U1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v   
```
// I ALSO HAVE A RAZER DEATHADDER ELITE HOOKED UP.
Bus 001 Device 004: ID 1532:024e Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x024e 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V3
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 243 lines cut ...]
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 Nov  2 18:02 .
drwxr-xr-x 4 root root 860 Nov  2 18:02 ..
lrwxrwxrwx 1 root root  10 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-event-if01 -> ../event11
lrwxrwxrwx 1 root root  10 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-event-if04 -> ../event15
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root  10 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_BlackWidow_V3-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_DeathAdder_Elite-event-mouse -> ../event3
lrwxrwxrwx 1 root root  10 Nov  2 18:02 usb-Razer_Razer_DeathAdder_Elite-if01-event-kbd -> ../event23
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_DeathAdder_Elite-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Nov  2 18:02 usb-Razer_Razer_DeathAdder_Elite-mouse -> ../mouse0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
When I have a little bit of time, maybe tomorrow I will attempt doing some packet captures.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
I will set up a Windows VM and attempt to gather this information as well.

justZaduma, 2020-11-03:
So the packet capture page says to toggle between 'Static' and 'Off' Well there is no Off options, so instead here is what I did. I went through every option under Quick Options in Synapse. Under many of the Quick Options there were settings. i.e. For 'Static' you had a 'Color Choice' For Stars you had Colors and Speed Choices.
I hope this is enough to go off of. If you need something else please let me know what to do. Also this pcapng file can't be shared on github for some reason? Here it is: [ufile.io external download link](https://ufile.io/wa3dhsn3)

Xanthous04, 2020-11-07:
are there any plans on getting razer-macos to have support withe BlackWidow v3? Really hoping to be able to customize the lighting effects really soon!

pieroxy, 2021-04-10:
Hello.

I have a few questions about this ticket:
1. Is it in progress and I should just wait until it is released?
2. If not, I was wondering if there is anything I can do to help on this topic. 

Please let me know if I can help in any way to move forward.

jake-arkinstall, 2021-04-21:
I'll second @pieroxy's comment. Anything we can do to help move this along would be great - we just need to know.

YoJames2019, 2021-04-23:
I'll third @pieroxy's comment.

Sprockettfuel, 2021-04-23:
Hi guys. I’m not ignoring this but I couldn’t find the thread on insider. Someone was able to help me. I recommended differentiating between the 0s and Os as I had to have a razer associate decode it for me

Heather L. Houpt

jake-arkinstall, 2021-05-06:
> Hi guys. I’m not ignoring this but I couldn’t find the thread on insider. Someone was able to help me. I recommended differentiating between the 0s and Os as I had to have a razer associate decode it for me

I'm confused.

Someone was able to help you do what?

FirephoenixX02, 2021-05-24:
So will there be support for the V3 or is it already supported? Im confused.

pieroxy, 2021-05-24:
> So will there be support for the V3 or is it already supported?

It is not yet supported that I am aware of.

FirephoenixX02, 2021-05-25:
> > So will there be support for the V3 or is it already supported?
> 
> It is not yet supported that I am aware of.

Thx for your Answer i wasn't sure if this issue was outdated or not.

Noah-Leick, 2021-06-13:
Figured I'd try my best to get pcapng files for this keyboard. Google Drive link contains an uncompressed folder and archives in 7z, xz, gz, and bz2 formats. Attached screenshot is of all pcapng files. Feel free to let me know if there is anything else I can do. [Drive link](https://drive.google.com/drive/folders/1lVXCzbew9f5v987D3OVdAdvNWqrLlezv?usp=sharing). 
[image]

prosaied, 2021-07-10:
@Noah-Leick Hi, hope you are doing well.
How use the files you have shared with us on your Google drive to install the Blackwidow V3 on Ubuntu Linux?

Spiffyk, 2021-08-04:
@prosaied These files are used as reference for the driver developers, you cannot use them directly to make BlackWidow V3 work. However, I have started work on this, as I have this keyboard too. You can watch #1572 to see if/when this work is done.

closed 2021-10-30 by z3ntu

pieroxy, 2021-11-30:
Hello. Any idea when is planned the next release (hopefully with this inside) ?

redhunglee, 2022-09-09:
I use RazerGenic to setup Blackwidow v3, just 4 key can not be light:
1. space key
2. right alt key ( when i light right alt key, Razer LOGO is light)
3. FN key
4. \ key

[image]

Spiffyk, 2022-09-09:
@redhunglee That is a problem on RazerGenie's side. Their editor is modeled after the Chroma keyboards, whose colour matrix mapping is slightly different from the BlackWidow. Polychromatic, another frontend for OpenRazer, has multiple modes in their editor - Elite would be the closest to BlackWidow.

redhunglee, 2022-09-10:
Thank you,I get it.
Polychromatic work fine.
I find how to custom single key by Polychromatic.