openrazer/openrazer #1322: Support for Razer BlackWidow V3 Pro

eeyrjmr, 2020-11-04:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-11-04:
Hi, please fill out the issue template that has been provided.

eeyrjmr, 2020-11-04:
uur sorry :( I had pasted the info but for some reason it wasn't added when I hit commit

z3ntu, 2020-11-04:
This is a duplicate of #1279 then.

closed 2020-11-04 by z3ntu

eeyrjmr, 2020-11-04:
While I agree this is a dupe, the "original" is missing the data

z3ntu, 2020-11-05:
I see everything from this issue in the other one as well - you just have to expand the sections to see it.

eeyrjmr, 2020-11-05:
you are right :(