openrazer/openrazer #1323: naga trinity does not save static color

flyer48, 2020-11-04:
Hi,

i use PopOS 20.10, latest openrazer is installed.

i have a Naga Trinity and it does not save the configured static colour, every time i reboot or waking my computer the colour has to be set again. The mouse always resets and breathes in different colours.

i also own a Tartarus v2 which works fine.

please help, thanks in advance.

lah7, 2020-11-05:
It's possible the firmware of this particular device defaults to breath between power cycles and expects software (i.e. usually Synapse 3) to control its state.

This should be addressed when #1149 is merged (so, possibly next release). In the meantime, you could rig up a script to run when you log on to your desktop session:

```
#!/usr/bin/python3
import openrazer.client
devices = openrazer.client.DeviceManager().devices
for device in devices:
    if device.name == "Razer Naga Trinity":
        device.fx.static(0,255,0)
```

:point_up: This sets a static green colour (0,255,0)

flyer48, 2020-11-05:
thanks for explaining the problem and giving me a solution, it works fine on startup and i am working on a wakeup solution, my scripting knowledge is very low...

nightsky30, 2021-01-17:
@flyer48 Would you mind testing this branch: https://github.com/nightsky30/openrazer/tree/fix_naga_trinity_static_effect to see if it fixes the static effect for you?

nightsky30, 2021-01-17:
@flyer48 Would you be willing to test this [branch](https://github.com/nightsky30/openrazer/tree/fix_naga_trinity_static_effect) to see if it fixes the static effect for you?

flyer48, 2021-01-17:
sorry, i don't have the mouse anymore. Thanks for your good work.

nightsky30, 2021-01-17:
@Black88ad Would you be willing to test this [branch](https://github.com/nightsky30/openrazer/tree/fix_naga_trinity_static_effect) to see if it fixes the static effect for you?

A similarly behaving device required these changes. Thought it might help here.

Black88ad, 2021-01-17:
@nightsky30  I have stop used Naga but i will try to check it. and will update You that it works or not. I have hope that it will be this week

TermoZour, 2021-01-18:
I can test the branch right now. How do I do it exactly? Clone your repo and install it as usual?

nightsky30, 2021-01-18:
@TermoZour That would be great! :smile: Yeah, I've got a branch here: https://github.com/nightsky30/openrazer/tree/fix_naga_trinity_static_effect Clone, checkout, build per usual.

If that works, I'd be interested in possibly getting the other effects working on it. Looks like it **should** support breathing, reactive, and spectrum as well.

TermoZour, 2021-01-18:
I have cloned the repo with your specified branch and ran the install script to generate the debs. 

I have installed said debs, rebooted the PC, and the mouse indeed restores the old color (at least from RazerGenie). It doesn't default to spectrum anymore.

nightsky30, 2021-01-18:
@TermoZour Nice! Does the color persist if you plug the mouse into a system not running openrazer (or Synapse)?

TermoZour, 2021-01-18:
No, it does not. It switches back to spectrum effect. Tested on a different laptop running Pop!_OS 20.10 who never saw openrazer.

nightsky30, 2021-01-18:
Ok, well that kills half that theory. :smile: Looks like the effect works without the need for the special report it's using now. But the effect isn't stored for whatever reason and the persistence added in the newer openrazer version is restoring it each time.

TermoZour, 2021-01-18:
At least the effect restores. That would also be nice for Razer Nari Ultimate since that one also resets the effect (static green). Too bad they don't have support for anything. I was working on it but couldn't get the driver working with those...

nightsky30, 2021-01-22:
@TermoZour I've updated that branch with some pretty bold changes based on the sort of "generation" of mouse the Trinity is along with the fact that other non-Lancehead mice have been shown to support separate zones including left/right sides.

These changes add the other missing effects, independent zone effects, independent zone brightness, suspend/resume lighting, left, logo, and scroll zones. If you would like to test and let me know what works and what doesn't I'd be very grateful.

I'm hopeful this will add a lot for this mouse.

TermoZour, 2021-01-23:
After pulling the branch, building all debs, installing all debs, restarting the PC, the daemon is not working.

RazerGenie is showing this:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2021-01-24 00:23:44 EET; 20ms ago
       Docs: man:openrazer-daemon(8)
    Process: 7029 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=0/SUCCESS)
   Main PID: 7029 (code=exited, status=0/SUCCESS)

Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:     super(RazerNagaTrinity, self).__init__(*args, **kwargs)
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 296, in __init__
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:     bright_func(self.zone[i]["brightness"])
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 125, in set_logo_brightness
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]:     with open(driver_path, 'w') as driver_file:
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]: PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0067.0001/logo_led_brightness'
Jan 24 00:23:44 termozour-pop-os openrazer-daemon[7029]: Stopping daemon.
Jan 24 00:23:44 termozour-pop-os systemd[2204]: openrazer-daemon.service: Succeeded.
```

Trying to run list_devices.py shows this:
```
termozour@termozour-pop-os:~/Repos/openrazer_nightsky-nari$ python3 ./examples/list_devices.py 
ERROR:dbus.proxies:Introspect error on :1.177:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
Traceback (most recent call last):
  File "./examples/list_devices.py", line 4, in <module>
    device_manager = DeviceManager()
  File "/usr/lib/python3/dist-packages/openrazer/client/__init__.py", line 32, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.177 was not provided by any .service files
```

Checking my user groups, I get this:
```
termozour@termozour-pop-os:~/Repos/openrazer_nightsky-nari$ groups termozour 
termozour : termozour adm disk sudo video plugdev vboxusers wireshark
```

nightsky30, 2021-01-24:
@TermoZour I committed some changes switching to the backlight led for the keypad. If it gives the same error, maybe delete the entry for the trinity from .config/openrazer/persistence.conf Thanks for testing this :)

TermoZour, 2021-01-25:
Same steps: git pull, build debs, install debs, reboot.

Opened RazerGenie:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Mon 2021-01-25 10:57:29 EET; 27ms ago
       Docs: man:openrazer-daemon(8)
    Process: 4605 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=0/SUCCESS)
   Main PID: 4605 (code=exited, status=0/SUCCESS)

Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     super(RazerNagaTrinity, self).__init__(*args, **kwargs)
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 296, in __init__
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     bright_func(self.zone[i]["brightness"])
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 125, in set_logo_brightness
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     with open(driver_path, 'w') as driver_file:
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]: PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0067.0001/logo_led_brightness'
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]: Stopping daemon.
Jan 25 10:57:29 termozour-pop-os systemd[1992]: openrazer-daemon.service: Succeeded.
```

Tried to list devices with python3 example script:
```
termozour@termozour-pop-os:~/Repos/openrazer_nightsky-nari$ python3 ./examples/list_devices.py 
ERROR:dbus.proxies:Introspect error on :1.162:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
Traceback (most recent call last):
  File "./examples/list_devices.py", line 4, in <module>
    device_manager = DeviceManager()
  File "/usr/lib/python3/dist-packages/openrazer/client/__init__.py", line 32, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.162 was not provided by any .service files
```

In `persistence.conf` I have a bunch of entries with UNKWN and 2 with actual serials. I deleted the one for the Trinity (I checked the back of the mouse), left the UNKWN untouched.
I rebooted and still got the same errors.

Just to try and rule out other issues, I completely removed openrazer and RazerGenie and reinstalled them from the official apt repo and they are back working so indeed there's something wrong with this branch/build.

nightsky30, 2021-01-25:
@TermoZour Roger that! I'll try and take a look at this later today. :+1:

TermoZour, 2021-01-25:
In the meantime, should I also try to remove the `persistence.conf` file completely and/or try to install your branch again?

nightsky30, 2021-01-25:
You could try if you want, but I'm thinking the issue lies elsewhere. Are those LEDs able to be set individually in Synapse, or are they all linked to one effect/color? Trying to determine if they all use the backlight/zero LED, or if they have separate LED zones (scroll/logo/left side). Was hoping they were separate. That might be the issue. But if they are separate, I'd like to implement it that way in openrazer so users have the option.

TermoZour, 2021-01-25:
They are separate in Synapse. Can it be the fact that I have attached the left side without lights?

nightsky30, 2021-01-26:
@TermoZour Thanks! That's helpful info. I've swapped the class the Naga Trinity extends to RazerDeviceBrightnessSuspend vs RazerDeviceSpecialBrightnessSuspend based on the error in device_base.py and the class the Naga Hex V2 is extending:

```
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     super(RazerNagaTrinity, self).__init__(*args, **kwargs)
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 25 10:57:29 termozour-pop-os openrazer-daemon[4605]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
```

Does this fix the daemon issue?

TermoZour, 2021-01-26:
No. Still throws errors

RazerGenie error:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Tue 2021-01-26 21:37:20 EET; 29ms ago
       Docs: man:openrazer-daemon(8)
    Process: 4691 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=0/SUCCESS)
   Main PID: 4691 (code=exited, status=0/SUCCESS)

Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 296, in __init__
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:     bright_func(self.zone[i]["brightness"])
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py", line 125, in set_logo_brightness
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]:     with open(driver_path, 'w') as driver_file:
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]: PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0067.0001/logo_led_brightness'
Jan 26 21:37:20 termozour-pop-os openrazer-daemon[4691]: Stopping daemon.
Jan 26 21:37:20 termozour-pop-os systemd[2041]: openrazer-daemon.service: Succeeded.
```

nightsky30, 2021-01-26:
Oh, that's interesting. Looks like RazerDeviceBrightnessSuspend extends RazerDeviceSpecialBrightnessSuspend and not the other way round. That explains the failure on the same line, but not WHY. Hmmm.

nightsky30, 2021-01-26:
Ok, I just commented out the suspend/resume to see if that will fix things for now and allow the testing of other stuff.

TermoZour, 2021-01-30:
@nightsky30 so I've tested your latest commit, RazerGenie doesn't crash anymore. First thing I noticed the 3 zones are there. The left one (detected as "Lighting" by RazerGenie) doesn't seem to work. The logo zone and the scroll wheel zone do work. Another issue is that when I set an effect for whichever zone, the other 2 go dead (no color). If I set the other two to whatever effect, only the last zone set will be shown on the mouse, the other 2 go dead (no color).

nightsky30, 2021-01-30:
> @nightsky30 so I've tested your latest commit, RazerGenie doesn't crash anymore. First thing I noticed the 3 zones are there. The left one (detected as "Lighting" by RazerGenie) doesn't seem to [...]

I bet the left zone probably works as I had it with the left LED value vs the backlight value/effects it is using now (just like the Left Handed Naga 2020 in PR now). Still no idea why the suspend/resume were causing the crashes.

That issue with the zones going dead is interesting too! I wonder if that has anything to do with that mode transition packet currently sent with static effects. The same is done for the charging pad chroma, but all 10 LEDs on that device are essentially one section. I just remember when dealing with the charging pad that sending that special command was essentially turning the LEDs off.

TermoZour, 2021-01-30:
That sure is interesting. I have no idea how you managed to figure all this stuff out but good job so far! If there's anything else I can help you with, just tag me. Unfortunately, I don't know where to start with reverse engineering the USB packets since I don't see any 'patterns' except the RGB hex values of _something_ in Wireshark. Whether that is a specific part of the mouse or any other device, idk.

nightsky30, 2021-01-31:
@TermoZour I reset to a previous commit that uses the left LED, rearranged the driver to model the Mamba Elite (like the Naga Left-handed Edition 2020), and I think I figured out what was causing the suspend/resume crashes.

Hopefully this works better :)

TermoZour, 2021-01-31:
The daemon crashes again.
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2021-01-31 10:22:00 EET; 30ms ago
       Docs: man:openrazer-daemon(8)
    Process: 4675 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=0/SUCCESS)
   Main PID: 4675 (code=exited, status=0/SUCCESS)

Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:     super(RazerNagaTrinity, self).__init__(*args, **kwargs)
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 296, in __init__
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:     bright_func(self.zone[i]["brightness"])
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/lanceheadte.py", line 88, in set_left_brightness
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]:     with open(driver_path, 'w') as driver_file:
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]: PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0067.0001/left_led_brightness'
Jan 31 10:22:00 termozour-pop-os openrazer-daemon[4675]: Stopping daemon.
Jan 31 10:22:00 termozour-pop-os systemd[2028]: openrazer-daemon.service: Succeeded.
```

nightsky30, 2021-01-31:
Ok, I commented out the suspend/resume again, but with all the other left LED changes in place.

Is that the entire error? It seems to be ok with the logo this time and threw an error on the left LED.

TermoZour, 2021-01-31:
This is the full error in RazerGenie (after just pulling latest changes):
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2021-01-31 15:55:06 EET; 29ms ago
       Docs: man:openrazer-daemon(8)
    Process: 6208 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=0/SUCCESS)
   Main PID: 6208 (code=exited, status=0/SUCCESS)

Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:     super(RazerNagaTrinity, self).__init__(*args, **kwargs)
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 976, in __init__
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:     super().__init__(device_path, device_number, config, persistence, testing, additional_interfaces, additional_methods)
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 296, in __init__
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:     bright_func(self.zone[i]["brightness"])
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/lanceheadte.py", line 88, in set_left_brightness
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]:     with open(driver_path, 'w') as driver_file:
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]: PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:1532:0067.0001/left_led_brightness'
Jan 31 15:55:06 termozour-pop-os openrazer-daemon[6208]: Stopping daemon.
Jan 31 15:55:06 termozour-pop-os systemd[2028]: openrazer-daemon.service: Succeeded.
```

ohaessler, 2021-06-16:
is there any progress on this?

closed 2021-11-16 by github-actions[bot]