openrazer/openrazer #1325: Mouse is not recognised

coocoorhin, 2020-11-08:
### Distribution

Ubuntu 20.04

### Support

- [ ] Have you tried rebooting?  YES
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? YES

I have 2 razer devices:
` 
~ lsusb | grep '1532:'
Bus 001 Device 003: ID 1532:0c01 Razer USA, Ltd Razer Goliathus Chroma
Bus 001 Device 006: ID 1532:0083 Razer USA, Ltd Razer Basilisk X HyperSpeed
`

but in polychromatic only the goliathus chroma is detected(and it is working flawlessly). the mouse Basilisk is not detected.

there is my log:
`
~ tail ~/.local/share/openrazer/logs/razer.log
2020-11-08 16:22:00 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-11-08 16:22:00 | razer                          | INFO     | Found device.0: 0003:1532:0C01.0002
2020-11-08 16:22:00 | razer.device0                  | INFO     | Initialising device.0 RazerGoliathus
2020-11-08 16:22:00 | razer                          | INFO     | Serving DBus
2020-11-08 16:28:26 | razer                          | INFO     | Stopping daemon.
2020-11-08 16:28:28 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 4846
2020-11-08 16:28:28 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-11-08 16:28:28 | razer                          | INFO     | Found device.0: 0003:1532:0C01.0002
2020-11-08 16:28:28 | razer.device0                  | INFO     | Initialising device.0 RazerGoliathus
2020-11-08 16:28:28 | razer                          | INFO     | Serving DBus

`

z3ntu, 2020-11-08:
v2.8.0 doesn't support your mouse yet, but it'll be included in the next release.

closed 2020-11-08 by z3ntu

z3ntu, 2020-11-09:
2.9.0 has been released yeterday. It's not available on Ubuntu yet though because launchpad has some internal issues. Hopefully it'll be available today or tomorrow.

coocoorhin, 2020-12-16:
After getting 2.9.0, the problem remains. The Basilisk X HyperSpeed is not recognized.
```
 ~ lsusb | grep '1532:'
Bus 001 Device 003: ID 1532:0c01 Razer USA, Ltd Razer Goliathus Chroma
Bus 001 Device 006: ID 1532:0083 Razer USA, Ltd Razer Basilisk X HyperSpeed
```
and log:
```
 ~ tail ~/.local/share/openrazer/logs/razer.log
2020-12-16 11:37:14 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-16 11:37:15 | razer                          | INFO     | Found device.0: 0003:1532:0C01.0002
2020-12-16 11:37:15 | razer.device0                  | INFO     | Initialising device.0 RazerGoliathus
2020-12-16 11:37:15 | razer                          | INFO     | Serving DBus
2020-12-16 11:43:29 | razer                          | INFO     | Stopping daemon.
2020-12-16 11:43:31 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 69041
2020-12-16 11:43:31 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-16 11:43:31 | razer                          | INFO     | Found device.0: 0003:1532:0C01.0002
2020-12-16 11:43:31 | razer.device0                  | INFO     | Initialising device.0 RazerGoliathus
2020-12-16 11:43:31 | razer                          | INFO     | Serving DBus
```