openrazer/openrazer #1327: Razer Blade 15 (Early 2020) Keyboard effects not working

maxkofler-dev, 2020-11-09:
Some effects on the Razer Blade 15 (Early 2020) do not work (Polychromatic says "Error in setting effect "..." on Razer Blade 15 (Early 2020)) also other Frontends don't are capable of setting these effects

z3ntu, 2020-11-09:
Hi, please provide the output of `lsusb | grep 1532` .
Additionally please start polychromatic-controller from the command line and check if you see an exception there when you get the "Error in setting effect" message in the UI.

maxkofler-dev, 2020-11-09:
> Hi, please provide the output of `lsusb | grep 1532` .
> Additionally please start polychromatic-controller from the command line and check if you see an exception there when you get the "Error in setting effect" message in the UI.

Ok, thankyou, I update the issue

z3ntu, 2020-11-11:
Does this happen on every boot or just sometimes? Maybe try rebooting a couple of times and checking if it doesn't work each time.

maxkofler, 2020-11-13:
> Does this happen on every boot or just sometimes? Maybe try rebooting a couple of times and checking if it doesn't work each time.

Nope, doesn't make a difference, also updated openrazer, it does not work...

closed 2021-06-20 by github-actions[bot]