openrazer/openrazer #1328: Support for RAZER KIYO

quantumfate, 2020-11-09:
### Device Info

* Name: Razer Kiyo
* Product Number: RZ19-02320100-R3M1
* Model Number: RZ19-0232

### Device Information
```
Bus 003 Device 004: ID 1532:0e03 Razer USA, Ltd Gaming Webcam [Kiyo]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0e03 Gaming Webcam [Kiyo]
  bcdDevice           79.13
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x02c7
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              0 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x0069
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bSourceID               2
[... 1132 lines cut ...]
    ls -lah /dev/input/by-id/

drwxr-xr-x 2 root root 320  9. Nov 14:00 .
drwxr-xr-x 4 root root 820  9. Nov 14:00 ..
lrwxrwxrwx 1 root root  11  9. Nov 14:00 usb-Alpha_Imaging_Tech._Corp._Razer_Kiyo-event-if00 -> ../event257
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Cooler_Master_Technology_Inc._AMD_SR4_lamplight_Control-event-if02 -> ../event25
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Cooler_Master_Technology_Inc._AMD_SR4_lamplight_Control-event-kbd -> ../event24
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Kingston_HyperX_Alloy_FPS_Mechanical_Gaming_Keyboard-event-if01 -> ../event30
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Kingston_HyperX_Alloy_FPS_Mechanical_Gaming_Keyboard-event-kbd -> ../event27
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Kingston_HyperX_Alloy_FPS_Mechanical_Gaming_Keyboard-if01-event-kbd -> ../event31
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Kingston_HyperX_Alloy_FPS_Mechanical_Gaming_Keyboard-if01-event-mouse -> ../event28
lrwxrwxrwx 1 root root   9  9. Nov 07:55 usb-Kingston_HyperX_Alloy_FPS_Mechanical_Gaming_Keyboard-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  11  9. Nov 07:55 usb-Kingston_HyperX_Cloud_Revolver_S_000000000000-event-if03 -> ../event256
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Razer_Razer_Naga_Trinity_00000000001A-event-if01 -> ../event21
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Razer_Razer_Naga_Trinity_00000000001A-event-mouse -> ../event18
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Razer_Razer_Naga_Trinity_00000000001A-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10  9. Nov 07:55 usb-Razer_Razer_Naga_Trinity_00000000001A-if02-event-kbd -> ../event23
lrwxrwxrwx 1 root root   9  9. Nov 07:55 usb-Razer_Razer_Naga_Trinity_00000000001A-mouse -> ../mouse0
```

### Packet Captures
I can provide information on the next weekend probably

### Synapse Screenshots
There is nothing

wallacelibrary, 2020-12-02:
Yes, please!

erickmathews, 2021-01-04:
Here are all of the screens for this in synapse.

[image]
[image]

Advanced Settings popup (This can also be spawned from OBS, and Windows)
[image]
[image]

kevchu3, 2021-02-19:
I recently purchased a Razer Kiyo as well, and I noticed we don't have OpenRazer support for it yet.  What do we need to move this forward?

quantumfate, 2021-02-22:
> I recently purchased a Razer Kiyo as well, and I noticed we don't have OpenRazer support for it yet. What do we need to move this forward?

pcap  is missing, can't do it right now

quantumfate, 2021-02-22:
> I recently purchased a Razer Kiyo as well, and I noticed we don't have OpenRazer support for it yet. What do we need to move this forward?

https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

This is what they are asking for

z3ntu, 2021-02-22:
Just FYI kernel support for the video capture is not in the scope of the openrazer project. You'll have to ask / develop support upstream in the kernel for that. OpenRazer is able to control additional functionality such as LED lights (which this device doesn't seem to have)

kevchu3, 2021-02-22:
I went ahead and set up Wireshark to capture the Razer Kiyo based on instructions @quantumfate provided in the link above.
lsusb: `Bus 001 Device 011: ID 1532:0e03 Razer USA, Ltd Gaming Webcam [Kiyo]`
Wireshark filter: `usb.bus_id == 1 && usb.device_address == 11`

 However, there were a couple things that didn't quite sit well with respect to functionality.

- Synapse allows the user to adjust the video (brightness, contrast, etc...) and the camera (zoom, focus, exposure, etc...).  However, as @z3ntu stated, this is outside of the scope of the OpenRazer project
- The ring light is physically adjustable when the video capture is on, and same with the LED indicator (light is on if video is being captured, off otherwise) but neither of these are adjustable from Synapse.

To prove this, I've captured a .pcapng where I have Synapse running, video on, and manually adjust the ring light's brightness on and off.

[razer_kiyo_pcapng.zip](https://github.com/openrazer/openrazer/files/6023837/razer_kiyo_pcapng.zip)

lachieh, 2021-02-25:
FWIW, the webcam settings are compatible with other applications like OBS, and on macOS "Webcam Settings" which means that it's probably an OS level or app-specific integration for webcam control.

The LED ring light is completely managed by the hardware only which means that it's never going to be possible to control via software — official or not.

As has been mentioned, managing the webcam settings settings is not in-scope for this project, so I would say that this issue is resolved.

rmakara, 2021-03-05:
When working on the support of Razer Kiyo maybe that would be possible to take a look at the brand new Razer Kiyo Pro. Fingers crossed. :)

lachieh, 2021-03-06:
Since there's no RGB lights at all on the Razer Kiyo Pro, it's highly unlikely.

z3ntu, 2021-03-13:
@kevchu3 I do not see any known packets in the capture, so as you said it doesn't seem that the lights are controllable via USB. So basically this device does not have anything that can be implemented in the scope of OpenRazer.

Can you all tell me what you're looking for with this camera? As far as I can find on the internet, the camera works fine out of the box on Linux, so I'm not sure why you're seeking openrazer support for it.

kevchu3, 2021-03-17:
@z3ntu After running through the Wireshark tests, I've concluded the same and would probably have this issue closed, if the rest of the folks agree

closed 2021-03-21 by z3ntu