openrazer/openrazer #1332: no volume wheel Razer BlackWidow Elite 1532:0228

sirstabbybeard, 2020-11-13:
hi i really could use some help.  i am running mint 20 cinnamon, fresh reinstall.  have a blackwidow elite keyboard i bought in early july.  at first the volume wheel was working fine when i was running mint 19.3.  i then tried to get an old windows media center remote control to work using rc_keymaps.  i never could quite get it to work right so any advise on that would be great, but i digress.  well after trying to get the remote to work my volume wheel stopped working and has not worked since. i started on mint 19.3 with this problem.  i have reinstalled both mint 19 and now mint 20, several times both with and without the remote connected the volume wheel still does not work.  I have second keyboard connected as well with a button volume that works fine.  the media center remote works the volume control also.  But again not the wheel on the keyboard.  when i use xev i do not register any key events when i move the wheel.  I assume it should be something like XF86AudiovolumeUP but nothing.  i also do not get any keyevents on several of the remote buttons either. i am hopeing someone out there might be able to help.  thanks in advance

sirstabbybeard, 2020-11-13:
please help i spent all day trying to fix this and have gotten nowhere.  i am really frustrated

z3ntu, 2020-11-17:
Hi, can you please try stopping the openrazer daemon and see if it works then? `systemctl --user stop openrazer-daemon.service`

felipemashki, 2021-02-23:
still having a problem?

closed 2021-07-24 by github-actions[bot]