openrazer/openrazer #1334: Mamba Elite stop taking commands

kblack227, 2020-11-15:
the mamba elite was working fine (except side lights) and stopped using static lighting that is currently set in the controller and when I try to adjust the dpi to anything other than 4500 and jumps right back. I have tried restarting the daemon and the pc. I have tried doing an apt-update to no avail. thanks for your help in advance. also, I have a black widow 2019 keyboard still working fine. this just started on 11-15-2020

z3ntu, 2020-11-17:
We've released a new openrazer version 9 days ago (on the 8th), I doubt your device has broken with that release though. You can try installing an v2.8.0 again and see if that helps.
Also have you changed anything else about your setup?

kblack227, 2020-11-17:
I figured it out, my 2-year-old pushed the profile button on the bottom of the mouse causing it to lose all of its settings. thanks for your help.

closed 2020-11-17 by kblack227