openrazer/openrazer #1335: Info / Docs for python API

Niwla23, 2020-11-17:
I read on your website that openrazer has a python API for creating custom effects, but I cant find any informations / docs / tutorials on the website, the README or the Wiki.

Is there any reference for it?

chabad360, 2020-11-17:
Unfortunately, the openrazer docs are not rendered to a nice format. You'll need to check the examples folder and the actual library code to understand what's available (fortunately, it's pretty simple).

Niwla23, 2020-11-17:
Okay, what about using google docstring style? It would allow building fancy docs with sphinx and host that on readthedocs.
If you dont want to stay with the current params / return scheme in docstrings I could make a PR with Sphinx config

Niwla23, 2020-11-17:
It would look something like this https://iservscrapping.readthedocs.io/en/latest/

Niwla23, 2020-11-17:
However I would wait anyways till after christmas when I hopefully have my new keyboard so I can actually test it

z3ntu, 2020-11-18:
We have some additional docs on the `feature_docs` branch and a rendered version is hosted at https://public.z3ntu.xyz/razer/pylib/ . I believe that some things are still undocumented though but it's probably easiest to look into the `examples/` directory of the repository or browse through the source of e.g. polychromatic.

closed 2021-06-20 by github-actions[bot]