openrazer/openrazer #1336: Missing Python libaries

EarlierPlane034, 2020-11-17:
Dependencies cannot be automatically installed on this distro.
Please see the README before using the application.
**************************************************
Checking if Razer modules are present...
**************************************************
The OpenRazer Python libraries are missing!
Please install them from http://openrazer.github.io/
They are expected to be installed here: /usr/lib/python3/dist-packages/razer/
**************************************************
Continue anyway? [y] 
im getting this error what should i do?

z3ntu, 2020-11-18:
Have you actually installed OpenRazer? The instructions can be found at https://openrazer.github.io/#download

lah7, 2020-11-18:
That looks like the `install.sh` script from Polychromatic's repository. I would recommend installing packages (for both projects) where available for your distribution. This `install.sh` script will become obsolete in future.

EarlierPlane034, 2020-11-19:
> That looks like the `install.sh` script from Polychromatic's repository. I would recommend installing packages (for both projects) where available for your distribution. This `install.sh` script will become obsolete in future.

I did I opened it from files what do I need to download?

EarlierPlane034, 2020-11-19:
> Have you actually installed OpenRazer? The instructions can be found at https://openrazer.github.io/#download

Yes I’m also on raspberrypi os which run Debian

lah7, 2020-11-20:
Please follow the Debian 10 ("Buster") instructions, like so:

Since you're running on a Raspberry Pi, you need to install `raspberrypi-kernel-headers` first and make sure the kernel is up-to-date too. This is so DKMS can install the OpenRazer driver against a matching kernel and its headers.

    sudo apt update
    sudo apt install raspberrypi-kernel-headers
    sudo apt upgrade raspberrypi-kernel

**OpenRazer**
```
echo 'deb http://download.opensuse.org/repositories/hardware:/razer/Debian_10/ /' | sudo tee /etc/apt/sources.list.d/hardware:razer.list
curl -fsSL https://download.opensuse.org/repositories/hardware:razer/Debian_10/Release.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/hardware_razer.gpg > /dev/null
sudo apt update
sudo apt install openrazer-meta
```
via https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta (which is linked on the OpenRazer website)

Then [install a frontend](https://github.com/openrazer/openrazer#applications), such as **Polychromatic** which you initially tried:

* https://polychromatic.app/download/debian/

Be sure to reboot the system so the driver can be initialised.

closed 2021-06-20 by github-actions[bot]