openrazer/openrazer #1337: Razer Cynosa V2 Volume Buttons

MCSH, 2020-11-18:
### Distribution

Arch Linux, razer version 2.9.0-1

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The volume up and volume down keys work as a down arrow and a left arrow respectively whenever the daemon is running. Killing the daemon fixes the issue, but that's not a real solution.

dsypniewski, 2020-12-02:
I'm having the same problem, before installation of this driver the keys were working just fine.

no95typem, 2020-12-05:
Manjaro. I am experiencing the same problem.

no95typem, 2020-12-15:
I experimented a little - the problem is that in the driver mode in the event files, the binary code for pressing the Volume + and Right Arrow (and Volume- and Down Arrow) keys are identical. I have not yet fully understood the driver code, and therefore I have a question: does this binary code in the event files depend only on the keyboard or on the driver on the PC too?

no95typem, 2020-12-23:
This should fix this #1385.

MCSH, 2020-12-23:
Thank you for solving it, @no95typem; I was planning on hacking away during the holidays, but I'm glad it's resolved (well, once they merge, it will be resolved, but we'll get there!)

closed 2020-12-27 by z3ntu

MCSH, 2020-12-27:
Any ideas when is the next release? Debating whether I should switch to git or remain on the previous release build.

z3ntu, 2020-12-27:
Definitely not very soon, so build the version you need for now.

commit 7e952111 (2020-12-27): Fix volume control buttons on Razer Cynosa V2

Fixes #1337