openrazer/openrazer #1339: Support for Razer Basilisk Hyperspeed X (Bluetooth)

szul, 2020-11-19:
Creating a new issue as per @bno1 here: https://github.com/openrazer/openrazer/issues/1307#issuecomment-729799427

The Razer Basilisk works fine via USB dongle, but not via Bluetooth.

Comments from @bno1 below:

> The driver doesn't work with the mouse in bluetooth mode and I'm not sure what needs to be done.
> 
> I found the device ID (`0082`) in `bluetoothctl`:
> 
> ```
> $ bluetoothctl info F2:12:E5:82:5A:B9
> Device F2:12:E5:82:5A:B9 (random)
>         Name: Basilisk X HyperSpeed
>         Alias: Basilisk X HyperSpeed
>         Appearance: 0x03c2
>         Icon: input-mouse
>         Paired: yes
>         Trusted: no
>         Blocked: no
>         Connected: yes
>         WakeAllowed: yes
>         LegacyPairing: no
>         UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
>         UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
>         UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
>         UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
>         UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
>         UUID: Vendor specific           (52401523-f97c-7f90-0e7f-6c6f4e36db1c)
>         Modalias: usb:v1532p0082d0001
>         Battery Percentage: 0x17 (23)
> ```
> 
> I tried adding `0082` to my build but the driver doesn't seem to catch it:
> 
> ```
> [  757.966862] input: Basilisk X Mouse as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input42
> [  757.967036] input: Basilisk X Consumer Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input43
> [  757.967120] input: Basilisk X System Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input44
> [  757.967197] input: Basilisk X as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input45
> [  757.967271] input: Basilisk X Keyboard as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input46
> [  757.967386] hid-generic 0005:1532:0082.0008: input,hidraw7: BLUETOOTH HID v0.01 Mouse [Basilisk X] on a8:7e:ea:d7:32:5e
> ```

Also, was asked to reference this comment for addition info: https://github.com/openrazer/openrazer/pull/1203#issuecomment-663992279

Chiqui1234, 2020-12-27:
+1 and support for Basilisk Ultimate and his dongle.
lsusb
`lsusb | grep Razer
Bus 001 Device 004: ID 1532:0088 Razer USA, Ltd Razer Basilisk Ultimate Dongle
Bus 001 Device 002: ID 1532:0086 Razer USA, Ltd Razer Basilisk Ultimate`

If somebody needs more info, just ask me! Thanks

WrathofthePast, 2021-12-20:
> Creating a new issue as per @bno1 here: [#1307 (comment)](https://github.com/openrazer/openrazer/issues/1307#issuecomment-729799427)
> 
> The Razer Basilisk works fine via USB dongle, but not via Bluetooth.
> 
> Comments from @bno1 below:
> 
> > The driver doesn't work with the mouse in bluetooth mode and I'm not sure what needs to be done.
> > I found the device ID (`0082`) in `bluetoothctl`:
> > ```
> > $ bluetoothctl info F2:12:E5:82:5A:B9
> > Device F2:12:E5:82:5A:B9 (random)
> >         Name: Basilisk X HyperSpeed
> >         Alias: Basilisk X HyperSpeed
> >         Appearance: 0x03c2
> >         Icon: input-mouse
> >         Paired: yes
> >         Trusted: no
> >         Blocked: no
> >         Connected: yes
> >         WakeAllowed: yes
> >         LegacyPairing: no
> >         UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
> >         UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
> >         UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
> >         UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
> >         UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
> >         UUID: Vendor specific           (52401523-f97c-7f90-0e7f-6c6f4e36db1c)
> >         Modalias: usb:v1532p0082d0001
> >         Battery Percentage: 0x17 (23)
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I tried adding `0082` to my build but the driver doesn't seem to catch it:
> > ```
> > [  757.966862] input: Basilisk X Mouse as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input42
> > [  757.967036] input: Basilisk X Consumer Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input43
> > [  757.967120] input: Basilisk X System Control as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input44
> > [  757.967197] input: Basilisk X as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input45
> > [  757.967271] input: Basilisk X Keyboard as /devices/virtual/misc/uhid/0005:1532:0082.0008/input/input46
> > [  757.967386] hid-generic 0005:1532:0082.0008: input,hidraw7: BLUETOOTH HID v0.01 Mouse [Basilisk X] on a8:7e:ea:d7:32:5e
> > ```
> 
> Also, was asked to reference this comment for addition info: [#1203 (comment)](https://github.com/openrazer/openrazer/pull/1203#issuecomment-663992279)

The bluetooth device-id is 0x0083, not 0x0082. That's your issue. [Source](https://github.com/1kc/razer-macos/issues/497#issuecomment-997467918)

szul, 2021-12-20:
@WrathofthePast This is an issue in the OpenRazer implementation? Is there something I can do on my end to rectify it?

WrathofthePast, 2021-12-20:
> @WrathofthePast This is an issue in the OpenRazer implementation? Is there something I can do on my end to rectify it?

Huh, this is interesting. openrazer put 0x0083 in the code. Is it working with a bluetooth connection as is? I'm confused why this issue is still open.

szul, 2021-12-20:
Doesn't recognize the Basilisk X Hyperspeed in bluetooth mode.

WrathofthePast, 2021-12-20:
This is downright bizarre. I've checked all the files and they all are correct at first glance. Since this uses Basilisk Ultimate Receiver for the receiver mode (meaning support wasn't needed for that), I'm confused how the PR for this was merged considering that it's not working. It's not like you were the only one saying that.

I have some ideas on what to check next, but I'll look at it later. The real test is if the port to macOS works (I'll try that later). If it does, that reduces the search area quite significantly from 7 to 4 files and the lines of code from +88 -1 to +18 -1 (assuming the issue isn't a typo in one of those three files).

bno1, 2021-12-20:
Hello. I haven't kept up to date with this project since adding support for the HyperSpeed X, but from what I remember this driver doesn't support bluetooth peripherals. At least I couldn't find any instance of someone mentioning that they work and nothing in the source code or docs to indicate bluetooth support.

WrathofthePast, 2022-01-26:
@bno1, I noticed that it wasn't added anywhere (only the receiver was). Do you think it'd work if it got an entry for bluetooth (different device-id than the receiver)? Also, I noticed that there seems to be a missing entry for the receiver in razermouse_driver.c for `get_battery`. It's present in `is_charging` though.

bno1, 2022-01-26:
> @bno1, I noticed that it wasn't added anywhere (only the receiver was). Do you think it'd work if it got an entry for bluetooth (different device-id than the receiver)? Also, I noticed that there [...]

I think I tried adding the device ID for bluetooth mode and it still wasn't working. I didn't look too much into it. If this works for other mice I don't see why it wouldn't work for this one.

About the `get_battery` method, I probably didn't add an entry there because it wasn't needed.

WrathofthePast, 2022-01-26:
> I think I tried adding the device ID for bluetooth mode and it still wasn't working. I didn't look too much into it. If this works for other mice I don't see why it wouldn't work for this one. [...]

I'm confused. The Basilisk X Hyperspeed doesn't support charging as it uses AAs. Furthermore, under `razer_mouse_probe` & `razer_mouse_disconnect`, it shows it supporting `charge_level`, `charge_status`, and `charge_low_threshold`, but yet it isn't listed under `get_battery`.

I feel as though it should be. The Basilisk X Hyperspeed is listed with the Atheris Receiver and Orochi V2 Bluetooth, and Orochi V2 Receiver everywhere except for `get_battery` where the Basilisk X Hyperspeed is absent. Furthermore, all of those devices also use AAs and do not support charging.

bno1, 2022-01-26:
The switch in `get_battery` just sets the transaction id for the command. I didn't add it there because the command works fine without setting the transaction id. The command is still sent to the mouse regardless if it's part of that switch or not.

Mice with AA battery have a mock `is_charging` because user-space applications try to read the `charge_status` property and crash if the device file for it is absent.

`charge_level` refers to the battery charge level, `charge_low_threshold` refers to the charge level threshold under which the mouse warns you that the battery is low by blinking its LED. Those are present on all wireless mice regardless of battery type.

WrathofthePast, 2022-01-27:
Thank you. That makes a lot more sense. Two more questions:

1. If you defined a device under `get_battery` with the wrong transaction id, what would happen?

> Mice with AA battery have a mock is_charging because user-space applications try to read the charge_status property and crash if the device file for it is absent.

2. Would said crash look like a blank window?

I'm working on porting this to macOS and I've the application window has been randomly blank, so this could explain it.

bno1, 2022-01-27:
>     1. If you defined a device under `get_battery` with the wrong transaction id, what would happen?

No idea, I don't remember what the transaction id is for. I would assume that the device would just ignore the command or return an error.

>     2. Would said crash look like a blank window?

That depends on the application. Some might throw the error in the terminal or logs and ignore it, others would fail to start or exit when trying to read the charge status. Here is the issue I opened on polychromatic when I discovered the problem: https://github.com/polychromatic/polychromatic/issues/246 (error happened at startup). Most user space apps probably use the openrazer python library, so they would fail similarly to polychromatic.

WrathofthePast, 2022-01-29:
Thank you. Another question: I see you didn't list the Basilisk X Hyperspeed under `razer_attr_read_poll_rate` and `razer_attr_write_poll_rate`. Does it not support multiple polling rates? I know it supports 1000Hz.

bno1, 2022-01-29:
@WrathofthePast It does. It's the same thing as `get_battery`. If you want to see what device files the driver supports look at `razer_mouse_probe`.

WrathofthePast, 2022-01-29:
> @WrathofthePast It does. It's the same thing as `get_battery`. If you want to see what device files the driver supports for a particular model look at `razer_mouse_probe`.

I saw that. Are you saying it's redundant? Can it be listed for `get_battery`, `read_poll_rate`, & `write_poll_rate` without being listed under `razer_mouse_probe`? I ask because the macOS port doesn't use `razer_mouse_probe`.

bno1, 2022-01-29:
What do you mean by "Can it be listed for get_battery, read_poll_rate, & write_poll_rate without being listed under razer_mouse_probe?". Those functions do completely different things, they're not comparable. I have no idea how macOS drivers work.

WrathofthePast, 2022-01-29:
> What do you mean by "Can it be listed for get_battery, read_poll_rate, & write_poll_rate without being listed under razer_mouse_probe?". Those functions do completely different things, they're not comparable. I have no idea how macOS drivers work.

You were mentioning the device files listed under `razer_mouse_probe`, so I asking if those are required for the features to work or if it can substituted under each individual feature.

bno1, 2022-01-29:
Like I said, the switch in `get_battery`, `read_poll_rate`, `write_poll_rate` only sets the command transaction id. It's not something that configures mouse features. Mouse features are configured in `razer_mouse_probe`.

WrathofthePast, 2022-01-29:
> Like I said, the switch in `get_battery`, `read_poll_rate`, `write_poll_rate` only sets the command transaction id. It's not something that configures mouse features. Mouse features are configured in `razer_mouse_probe`.

I see. That makes much more sense. I guess what I was asking was whether it needed a transaction.id set, but I guess not. Thanks for clarifying this. 

On a side note, I noticed Orochi V2 Bluetooth is listed in the drivers. Does that work? If so, do you think it's possible to get bluetooth working for the Basilisk X Hyperspeed.

bno1, 2022-01-29:
Idk, someone has to try. I'm not going to put more effort into this because I don't have the time and I replaced my mouse because it had some issues.

kevinquillen, 2023-01-17:
So is this still not supported in Bluetooth mode?

PatalJunior, 2023-05-14:
Just a thing to throw here, it's possible to view the exact percentage of battery in bluetooth (at least on windows), so I was wondering if it would be possible to get the battery percentage thru the dongle as well.

biohazard-dayz, 2023-06-14:
Not working under Bluetooth still? I've been trying everything..