openrazer/openrazer #1343: Device not recognized

66Gramms, 2020-11-21:
#467 # Manjaro

### Support

- [ yes ] Have you tried rebooting?
- [ yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Polychromatic Controller says that it detects a razer device but daemon can't. My device is a razer ornata chroma which I checked and it is among the supported devices.

I have openrazer-daemon 2.9.0-1 installed from pamac and after reading the troubleshooting page and trying the commands described there I got the following:
```
sudo dkms install openrazer-driver/2.9.0-1
[sudo] password for gramms: 
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-2.9.0-1 does not exist.
```
What might be the problem and how could I resolve it?

z3ntu, 2020-11-22:
Please try `sudo dkms install openrazer-driver/2.9.0` without the `-1`

66Gramms, 2020-11-22:
Thank you, that did install it but now I get the following:
```
Your kernel headers for kernel 5.8.18-1-MANJARO cannot be found at
/usr/lib/modules/5.8.18-1-MANJARO/build or /usr/lib/modules/5.8.18-1-MANJARO/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
```

z3ntu, 2020-11-22:
Then please install the kernel headers for you kernel (i guess `linux58-headers`?), otherwise dkms can't work correctly.

66Gramms, 2020-11-22:
To be honest I wasn't sure about what kernel headers are and I couldn't find what should I install but I found it with `linux58-headers` so thank you :) after installing it and rebooting it works now!

closed 2020-11-22 by 66Gramms