openrazer/openrazer #1344: Razerkbd nowhere to be found in manjaro

TcdUqkZZZ, 2020-11-21:
### Distribution

### Support

after installing, both from the AUR openrazer-meta and openrazer-meta-git packages, there is no razerkbd  under /sys/bus/hid/drivers/, nor does `lsmod | grep razerkbd` return anything.

daemon runs, but obviously cannot bind the device, so client software doesn't recognize it. `razer_mount log: sudo journalctl -b | grep -i razer_mount` similarly returns nothing, despite razer_mount script actually being where it should be.

as far as the deamon goes, it starts correctly and then remains at SERVING DBUS.

Thanks for the help.

TcdUqkZZZ, 2020-11-21:
Uodate: after solving this issue, I still find the drivers not working: even tryingto bind the device as root results in failure with error no such device, while binding and unbinding in hid-generic works perfectly.

z3ntu, 2020-11-22:
Can you please post the output of the command `lsusb | grep 1532` ?

googleboy360, 2021-01-13:
any update on this just got my keyboard today! trying to figure out a way now

yumyumfarted, 2021-03-31:
Hello, 
I'm having a similar issue. I'll attach the files and output that I'm having:

(I'm using Poly-chromatic, so I'll attach a screenshot of the troubleshooter [https://imgur.com/a/KaI6nLv](here))

After running "sudo modprobe razerkbd" manually, I got this output.

'''
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.9.16-1-MANJARO

'''
I checked and all of my packages are built to py 3.9 as well. I couldn't tell if I had to rebuild everything to 3.8, or rebuild to 3.9. If it's the 3.9, I won't have to do anything.

z3ntu, 2021-03-31:
@yumyumfarted Try `sudo dkms install openrazer-driver/3.0.0`. You're likely missing the kernel headers for your kernel (`linux59-headers`)

yumyumfarted, 2021-03-31:
Okay, the dkms module is loaded now. If you look at the picture, there are only two errors now. "DKMS module is currently loaded" and "Check for supported hardware". Which I have checked and it is supported.

closed 2021-08-30 by github-actions[bot]