openrazer/openrazer #1345: "OpenRazer daemon is not installed"

Motzumoto, 2020-11-21:
### Distribution
Manjaro XFCE 
kernel: linux 54

### Support
- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Ive looked online and through the troubleshooting page countless times, all saying basically the same things, i ran the devplug thing, rebooted, still had the same error, i uninstalled, reinstalled uninstalled and reinstalled countless times and that also didnt work, im at a loss here and no one seems to have the same issue as me or they're able to fix it without giving the steps on how to do that.

Speaking of steps, im not an expert programmer, so baby steps would be really needed.

z3ntu, 2020-11-22:
Can you please run the following commands and post the output of each?
* `sudo dkms install openrazer-driver/2.9.0`
* `sudo modprobe razerkbd`
* `openrazer-daemon -Fv`

timofeika, 2020-12-05:
> Can you please run the following commands and post the output of each?
> 
> * `sudo dkms install openrazer-driver/2.9.0`
> * `sudo modprobe razerkbd`
> * `openrazer-daemon -Fv`

timka@tim-arch ~ % sudo dkms install openrazer-driver/2.9.0
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-2.9.0 does not exist.
timka@tim-arch ~ % sudo modprobe razerkbd                  
timka@tim-arch ~ % openrazer-daemon -Fv                    
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'

lah7, 2020-12-05:
@timofeika Assuming you're on Arch/Manjaro too, make sure you have the correct Linux headers package installed (for instance, if the output of `uname -r` starts with `5.4.` then install `linux54-headers`.

Then, please reinstall `openrazer-daemon`, `openrazer-driver-dkms` and `python-openrazer`from the AUR.

StudenToo925, 2020-12-09:
I seem to have similiar issue, but  I 1) always had linux headers installed 2) had everything working for a few days

output of 
   

- sudo dkms install openrazer-driver/2.9.0

   

-  sudo modprobe razerkbd

    

- openrazer-daemon -Fv

[studentoo925@studentoo925-pc ~]$ sudo dkms install openrazer-driver/2.9.0
Module openrazer-driver/2.9.0 already installed on kernel 5.9.12-1-MANJARO/x86_64
[studentoo925@studentoo925-pc ~]$ sudo modprobe razerkbd
[studentoo925@studentoo925-pc ~]$ openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'

z3ntu, 2020-12-09:
Please see the pinned issue: https://github.com/openrazer/openrazer/issues/1362

timofeika, 2020-12-18:
> Please see the pinned issue: #1362

This worked, thanks!

closed 2021-05-20 by z3ntu