openrazer/openrazer #1346: Razer Viper RGB

Aeirex, 2020-11-22:
The modes for color do not work properly, so it will not allow me to change the color at all on the mouse. I have to go to macOS and do it, but RGB requires polling to the mouse constantly which requires the OpenRazer daemon to do so, which it doesn't seem to work for me.

### Distribution
Fedora 33

### Support
The RGB does not work with the device when using GUI like Polychromatic, it works on macOS for me (using Razer Synapse). Something to do with the daemon not being able to set states with the mouse properly?

z3ntu, 2020-11-22:
Hi are you talking about the device `1532:0078` ? You can check with `lsusb | grep 1532` output.

Aeirex, 2020-11-28:
Bus 001 Device 008: ID 1532:0078 Razer USA, Ltd Viper (wired)

z3ntu, 2020-11-29:
@foxandr Does your Viper still work properly with the latest openrazer version?

closed 2021-06-20 by github-actions[bot]