openrazer/openrazer #1348: Kernel compatibility

RiccardoBarbieri, 2020-11-25:
Is OpenRazer compatible with linux kernel 5.10?
If not when will it be?

z3ntu, 2020-11-25:
Hi, we haven't had any compatibility issues for many many kernel versions so I doubt 5.10 will be a problem but I haven't explicitly tried it yet because 5.10 isn't out yet. Why are you asking?

RiccardoBarbieri, 2020-11-25:
@z3ntu I'm asking because Ihave the 5.10 (for compatibility with my mobo lan) and it says that kernel headers 5.10 are no supported.

z3ntu, 2020-11-25:
Can you get me the exact error you're seeing? Generally you need the kernel headers for your exact kernel version installed, it sounds like you might not have done that.

IreIsaac, 2020-12-15:
I am running into this issue, or at least a similar one.  Below is the error i am seeing plus kernel and OS version. @z3ntu Let me know if you have a work around/fix, please.

```
pop-os:~$ sudo apt install openrazer-meta 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  openrazer-daemon openrazer-driver-dkms python3-openrazer
Suggested packages:
  openrazer-doc
The following NEW packages will be installed:
  openrazer-daemon openrazer-driver-dkms openrazer-meta python3-openrazer
0 upgraded, 4 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/152 kB of archives.
After this operation, 1,169 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package openrazer-driver-dkms.
(Reading database ... 295625 files and directories currently installed.)
Preparing to unpack .../openrazer-driver-dkms_2.9.0-1~585~ubuntu20.10.1_all.deb ...
Unpacking openrazer-driver-dkms (2.9.0-1~585~ubuntu20.10.1) ...
Selecting previously unselected package openrazer-daemon.
Preparing to unpack .../openrazer-daemon_2.9.0-1~585~ubuntu20.10.1_all.deb ...
Unpacking openrazer-daemon (2.9.0-1~585~ubuntu20.10.1) ...
Selecting previously unselected package python3-openrazer.
Preparing to unpack .../python3-openrazer_2.9.0-1~585~ubuntu20.10.1_all.deb ...
Unpacking python3-openrazer (2.9.0-1~585~ubuntu20.10.1) ...
Selecting previously unselected package openrazer-meta.
Preparing to unpack .../openrazer-meta_2.9.0-1~585~ubuntu20.10.1_all.deb ...
Unpacking openrazer-meta (2.9.0-1~585~ubuntu20.10.1) ...
Setting up openrazer-driver-dkms (2.9.0-1~585~ubuntu20.10.1) ...
Loading new openrazer-driver-2.9.0 DKMS files...
Building for 5.10.1-051001-generic
Building initial module for 5.10.1-051001-generic
ERROR (dkms apport): kernel package linux-headers-5.10.1-051001-generic is not supported
Error! Bad return status for module build on kernel: 5.10.1-051001-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/2.9.0/build/make.log for more information.
dpkg: error processing package openrazer-driver-dkms (--configure):
 installed openrazer-driver-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 2.9.0-1~585~ubuntu20.10.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-meta (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on openrazer-driver-dkms (= 2.9.0-1~585~ubuntu20.10.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-daemon (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 2.9.0-1~585~ubuntu20.10.1); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package python3-openrazer (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup error from a previous failure.
                                                                                                          No apport report written because the error message indicates its a followup error fr
om a previous failure.
                      No apport report written because MaxReports is reached already
                                                                                    Processing triggers for man-db (2.9.3-2) ...
Errors were encountered while processing:
 openrazer-driver-dkms
 openrazer-meta
 openrazer-daemon
 python3-openrazer
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
```
pop-os:~$ uname -a
Linux pop-os 5.10.1-051001-generic #202012142031 SMP Mon Dec 14 20:36:58 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
```
pop-os:~$ cat /etc/os-release
NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os
```

lah7, 2020-12-16:
@IreIsaac  Please could you paste the output of `/var/lib/dkms/openrazer-driver/2.9.0/build/make.log`?

IreIsaac, 2020-12-16:
@lah7 here is what `/var/lib/dkms/openrazer-driver/2.9.0/build/make.log` shows right after attempting to `apt install` 

```
root@pop-os:/proc# cat /var/lib/dkms/openrazer-driver/2.9.0/build/make.log 
DKMS make.log for openrazer-driver-2.9.0 for kernel 5.10.1-051001-generic (x86_64)
Wed Dec 16 09:42:31 AM EST 2020
-e 
:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/5.10.1-051001-generic/build M=/var/lib/dkms/openrazer-driver/2.9.0/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.1-051001-generic'
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razerkbd_driver.o
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razercommon.o
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razerchromacommon.o
  LD [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razerkbd.o
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razermouse_driver.o
  LD [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razermouse.o
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razerkraken_driver.o
  LD [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razerkraken.o
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razeraccessory_driver.o
  LD [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razeraccessory.o
  MODPOST /var/lib/dkms/openrazer-driver/2.9.0/build/driver/Module.symvers
  CC [M]  /var/lib/dkms/openrazer-driver/2.9.0/build/driver/razeraccessory.mod.o
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/openrazer-driver/2.9.0/build/driver/razeraccessory.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1709: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.1-051001-generic'
make: *** [Makefile:25: driver] Error 2

```

z3ntu, 2020-12-16:
Looks like https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131 which has supposedly been fixed 2 weeks ago.

IreIsaac, 2020-12-16:
I am on the latest kernel, [v5.10.1](https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.10.1/amd64/), maybe its a Pop_OS thing and I need to wait for an update from pop? I'll keep looking into that bug/patch, see if i can hack something together. 

PS, I get same error when i build .deb's from the repo. 

I really appreciate you looking into this, and the responses, thanks!

z3ntu, 2020-12-18:
Maybe you should not use this repository at all though

> The mainline kernel builds are produced for debugging purposes and therefore come with no support. Use them at your own risk.

turin331, 2021-01-03:
> Maybe you should not use this repository at all though
> 
> > The mainline kernel builds are produced for debugging purposes and therefore come with no support. Use them at your own risk.

This repo is just mirroring the mainline kernel. There are similar issues with 5.10.1 in a while now. Its not a repo issue.  

That being said the bug is fixed in the next point releases. I am on 5.01.4 and the deamon runs fine.

QuantumTarantino, 2021-01-15:
Can't use virtualbox after installing the driver on the system. Dkms seems broken. I tried to reinstall the virtualbox dkms but always shows the same issue.

**Error! Bad return status for module build on kernel: 5.8.0-38-generic (x86_64)**

closed 2021-06-20 by github-actions[bot]